vmatare/thinkfan #2: ERROR: /proc/acpi/ibm/thermal: basic_ios::clear: Success

kyrias, 2015-11-25:
Just tried the new thinkfan on my L420, and as soon as I try to run it I get 

```ERROR: /proc/acpi/ibm/thermal: basic_ios::clear: Success```

Nothing else is printed, and then it exits.

vmatare, 2015-11-25:
Thanks a lot for the report! Are you sure that `/proc/acpi/ibm/thermal` actually exists? Current distributions don't have it any more, they put the thinkpad sensors into the more conformant path `/sys/devices/platform/thinkpad_hwmon/hwmon/*`.

kyrias, 2015-11-25:
Positive.

vmatare, 2015-11-25:
Oh, I got it. I'm not parsing the file correctly since in fact I don't even know what it's supposed to look like. Could you post the contents of your `/proc/acpi/ibm/thermal` so I can fix the parser?

kyrias, 2015-11-25:
Mine looks like this:

```
temperatures:	46 0 46 0 34 58 29 58
```

Though the number of temperatures will of course vary.

kyrias, 2015-11-25:
Another bug when trying to use the pwm interface:

```
ERROR: set_speed: Writing to /sys/class/hwmon/hwmon1/device/pwm1: Success
```

---

Also, time to update the README? ;)

```
ERROR: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!
```

vmatare, 2015-11-25:
Ah well. std::ios_base error handling is a horrible mess. If gcc at least implemented C++11 correctly... But for now I fear we'll have to live with these braindead error messages. Anyways, fix(es) are in the making.

vmatare, 2015-11-25:
Thanks again for the report, this actually revealed a whole bunch of bugs in the file I/O and related error handling. Try again with the updated master, it should give much better error messages and besides fix your issues ;-)

kyrias, 2015-11-25:
Now it prints

```
ERROR: /proc/acpi/ibm/thermal: Unknown file format.
```

instead ;p

vmatare, 2015-11-26:
Though you have to admit that the error message has improved significantly. Even better, it should be fixed for real now (cf. 6706cf4a)

kyrias, 2015-11-26:
Much better, yes. Don't have time to test it right now, but will later.

I am however getting occasional coredumps. [Here](https://theos.kyriasis.com/~kyrias/thinkfan.core) is one of them.

vmatare, 2015-11-26:
Turns out these core dumps are utterly useless since I'd need your binary along with it. I'm currently rewriting the whole crash handling so that it simply produces stack traces. It would be great if you could open a new issue with some more information about these crashes, like the config you're using,  the exact console output and maybe other steps needed to reproduce.

closed 2015-12-11 by vmatare