vmatare/thinkfan #4: weird issue

sixsixfive, 2015-12-13:
can't launch it

http://pastebin.com/RrZ6JAam

same happens if I just run /usr/bin/thinkfan as root

also my /proc/acpi/ibm/thermal
```
[localhost hj]# cat /proc/acpi/ibm/thermal
temperatures:   52 0 52 0 0 0 35 0
```

vmatare, 2015-12-14:
897dde3 should fix this, but I can't test it since my system doesn't have /proc/acpi/ibm/thermal any more. Please let me know wether it works with the updated master.

sixsixfive, 2015-12-14:
yes that works but now it hangs(I guess it doesn't give a proper status back to the shell)

eg after a start

```
[hj@localhost ~]$ rc-status
Runlevel: default
 dbus                                                                                     [  started  ]
 bluetooth                                                                                [  started  ]
 thinkfan                                                                                 [ starting  ]
 syslog-ng                                                                                [  started  ]
 NetworkManager                                                                           [  started  ]
 cupsd                                                                                    [  started  ]
 consolekit                                                                               [  started  ]
 netmount                                                                                 [  started  ]
 xdm                                                                                      [  started  ]
 acpid                                                                                    [  started  ]
 cronie                                                                                   [  started  ]
 alsasound                                                                                [  started  ]```

if I add an '&' to the init file it will also be started eg.

```
-start-stop-daemon --start --exec /usr/sbin/thinkfan -- -q -s5 -c /etc/thinkfan.conf
+start-stop-daemon --start --exec /usr/sbin/thinkfan -- -q -s5 -c /etc/thinkfan.conf &
```

vmatare, 2015-12-14:
Oh yeah that's right. The old versions used to daemonize by default, but the newer version doesn't daemonize at all. That's why you have to put the & at the end, to make the shell daemonize it. But thanks for reporting this issue. I'll see about implementing daemon mode again.

closed 2015-12-14 by vmatare