vmatare/thinkfan #5: thinkfan crashes (double free or corruption)

sixsixfive, 2015-12-14:
well ok it works now but it is crashing after some time ~ 10 minutes

```
[localhost hj]# rc-service thinkfan start
 * Starting thinkfan ...

WARNING: Using default fan control in /proc/acpi/ibm/fan.
Using default temperature inputs in /proc/acpi/ibm/thermal.

sleeptime=5, tmax=64, last_tmax=-128, biased_tmax=64 -> fan="level 4"
sleeptime=5, tmax=65, last_tmax=64, biased_tmax=65 -> fan="level 5"
sleeptime=5, tmax=66, last_tmax=65, biased_tmax=66 -> fan="level 7"
sleeptime=5, tmax=62, last_tmax=65, biased_tmax=62 -> fan="level 5"
sleeptime=5, tmax=66, last_tmax=65, biased_tmax=66 -> fan="level 7"
sleeptime=5, tmax=60, last_tmax=63, biased_tmax=60 -> fan="level 5"
sleeptime=5, tmax=58, last_tmax=59, biased_tmax=58 -> fan="level 4"
sleeptime=5, tmax=55, last_tmax=56, biased_tmax=55 -> fan="level 3"
sleeptime=5, tmax=79, last_tmax=55, biased_tmax=79 -> fan="level 7"
sleeptime=5, tmax=60, last_tmax=79, biased_tmax=60 -> fan="level 5"
sleeptime=2, tmax=120, last_tmax=60, biased_tmax=120 -> fan="level 7"
sleeptime=5, tmax=63, last_tmax=120, biased_tmax=-87 -> fan="level 0"
sleeptime=2, tmax=819, last_tmax=63, biased_tmax=819 -> fan="level 7"
sleeptime=5, tmax=59, last_tmax=4320, biased_tmax=59 -> fan="level 5"
sleeptime=2, tmax=25920, last_tmax=59, biased_tmax=25920 -> fan="level 7"
sleeptime=5, tmax=56, last_tmax=25920, biased_tmax=56 -> fan="level 4"
*** Error in `/usr/sbin/thinkfan': double free or corruption (fasttop): 0x000000000214ff60 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x72055)[0x7f778b827055]
/usr/lib/libc.so.6(+0x779a6)[0x7f778b82c9a6]
/usr/lib/libc.so.6(+0x7818e)[0x7f778b82d18e]
/usr/sbin/thinkfan(_ZNKSt14default_deleteIN8thinkfan16TemperatureStateEEclEPS1_+0x2a)[0x430e62]
/usr/sbin/thinkfan(_ZNSt10unique_ptrIN8thinkfan16TemperatureStateESt14default_deleteIS1_EED1Ev+0x47)[0x4302d5]
/usr/sbin/thinkfan(_ZN8thinkfan3runERKNS_6ConfigE+0xbcc)[0x42dccb]
/usr/sbin/thinkfan(main+0x226)[0x42f250]
/usr/lib/libc.so.6(__libc_start_main+0xf0)[0x7f778b7d5610]
/usr/sbin/thinkfan(_start+0x29)[0x42cd09]
======= Memory map: ========
00400000-00455000 r-xp 00000000 08:12 286290                             /usr/bin/thinkfan
00655000-00656000 r--p 00055000 08:12 286290                             /usr/bin/thinkfan
00656000-00657000 rw-p 00056000 08:12 286290                             /usr/bin/thinkfan
0213c000-0216e000 rw-p 00000000 00:00 0                                  [heap]
7f778b291000-7f778b38e000 r-xp 00000000 08:12 267493                     /usr/lib/libm-2.22.so
7f778b38e000-7f778b58d000 ---p 000fd000 08:12 267493                     /usr/lib/libm-2.22.so
7f778b58d000-7f778b58e000 r--p 000fc000 08:12 267493                     /usr/lib/libm-2.22.so
7f778b58e000-7f778b58f000 rw-p 000fd000 08:12 267493                     /usr/lib/libm-2.22.so
7f778b58f000-7f778b5b4000 r-xp 00000000 08:12 264257                     /usr/lib/libudev.so.1.6.3
7f778b5b4000-7f778b7b3000 ---p 00025000 08:12 264257                     /usr/lib/libudev.so.1.6.3
7f778b7b3000-7f778b7b4000 r--p 00024000 08:12 264257                     /usr/lib/libudev.so.1.6.3
7f778b7b4000-7f778b7b5000 rw-p 00025000 08:12 264257                     /usr/lib/libudev.so.1.6.3
7f778b7b5000-7f778b950000 r-xp 00000000 08:12 267149                     /usr/lib/libc-2.22.so
7f778b950000-7f778bb4f000 ---p 0019b000 08:12 267149                     /usr/lib/libc-2.22.so
7f778bb4f000-7f778bb53000 r--p 0019a000 08:12 267149                     /usr/lib/libc-2.22.so
7f778bb53000-7f778bb55000 rw-p 0019e000 08:12 267149                     /usr/lib/libc-2.22.so
7f778bb55000-7f778bb59000 rw-p 00000000 00:00 0 
7f778bb59000-7f778bb6f000 r-xp 00000000 08:12 267315                     /usr/lib/libgcc_s.so.1
7f778bb6f000-7f778bd6e000 ---p 00016000 08:12 267315                     /usr/lib/libgcc_s.so.1
7f778bd6e000-7f778bd6f000 rw-p 00015000 08:12 267315                     /usr/lib/libgcc_s.so.1
7f778bd6f000-7f778bee1000 r-xp 00000000 08:12 267748                     /usr/lib/libstdc++.so.6.0.21
7f778bee1000-7f778c0e1000 ---p 00172000 08:12 267748                     /usr/lib/libstdc++.so.6.0.21
7f778c0e1000-7f778c0eb000 r--p 00172000 08:12 267748                     /usr/lib/libstdc++.so.6.0.21
7f778c0eb000-7f778c0ed000 rw-p 0017c000 08:12 267748                     /usr/lib/libstdc++.so.6.0.21
7f778c0ed000-7f778c0f1000 rw-p 00000000 00:00 0 
7f778c0f1000-7f778c0fc000 r-xp 00000000 08:12 281886                     /usr/lib/libatasmart.so.4.0.5
7f778c0fc000-7f778c2fc000 ---p 0000b000 08:12 281886                     /usr/lib/libatasmart.so.4.0.5
7f778c2fc000-7f778c2fe000 r--p 0000b000 08:12 281886                     /usr/lib/libatasmart.so.4.0.5
7f778c2fe000-7f778c2ff000 rw-p 0000d000 08:12 281886                     /usr/lib/libatasmart.so.4.0.5
7f778c2ff000-7f778c321000 r-xp 00000000 08:12 267050                     /usr/lib/ld-2.22.so
7f778c4f4000-7f778c4fb000 rw-p 00000000 00:00 0 
7f778c51f000-7f778c520000 rw-p 00000000 00:00 0 
7f778c520000-7f778c521000 r--p 00021000 08:12 267050                     /usr/lib/ld-2.22.so
7f778c521000-7f778c522000 rw-p 00022000 08:12 267050                     /usr/lib/ld-2.22.so
7f778c522000-7f778c523000 rw-p 00000000 00:00 0 
7ffe95360000-7ffe95381000 rw-p 00000000 00:00 0                          [stack]
7ffe953ea000-7ffe953ec000 r--p 00000000 00:00 0                          [vvar]
7ffe953ec000-7ffe953ee000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
 * start-stop-daemon: failed to start `/usr/sbin/thinkfan'                                       [ !! ]
 * ERROR: thinkfan failed to start
[localhost hj]# 
```

vmatare, 2015-12-14:
That was a really dumb typo. Sorry. Fixed in 6249d81.
However what's happening here is that your service manager is terminating thinkfan after ~10 minutes because it's not backgrounding. The terminate signal then triggers the double free error. So I guess I do have to implement daemon mode ;-)
Thanks for reporting!

sixsixfive, 2015-12-14:
ok, also there is another thing thinkfan doesnt create a *.pid(needed for the reload in init script)

vmatare, 2015-12-14:
yup, will do that along with the daemonization

closed 2015-12-14 by vmatare