vmatare/thinkfan #7: ERROR: Segmentation fault.

sixsixfive, 2015-12-15:
;) I tried this now a few times always with the same result

```
Dec 15 04:18:26 localhost thinkfan[2635]: Using default fan control in /proc/acpi/ibm/fan.
Dec 15 04:18:26 localhost thinkfan[2635]: Using default temperature inputs in /proc/acpi/ibm/thermal.
Dec 15 04:18:26 localhost thinkfan[2635]: sleeptime=5, tmax=67, last_tmax=-128, biased_tmax=67 -> fan="level 7"
Dec 15 04:34:21 localhost thinkfan[2635]: sleeptime=5, tmax=61, last_tmax=90, biased_tmax=-51 -> fan="level 7"
Dec 15 04:34:56 localhost thinkfan[2635]: sleeptime=2, tmax=116640, last_tmax=62, biased_tmax=116640 -> fan="
Dec 15 04:34:56 localhost thinkfan[2635]: ERROR: Segmentation fault.
Dec 15 04:34:56 localhost thinkfan[2635]: Backtrace:
Dec 15 04:34:56 localhost thinkfan[2635]: thinkfan() [0x447ebf]
thinkfan(_ZN8thinkfan5ErrorC2ERKSs+0x5f) [0x44802d]
thinkfan(_ZN8thinkfan3BugC1ERKSs+0x23) [0x4480cd]
thinkfan(_ZN8thinkfan11sig_handlerEi+0xd5) [0x42ddd8]
thinkfan(_ZN8thinkfan6LoggerlsERKSs+0x27) [0x440801]
thinkfan(_ZN8thinkfan3runERKNS_6ConfigE+0x71d) [0x42e58c]
thinkfan(main+0x418) [0x430488]
thinkfan(_start+0x29) [0x42da79]
Dec 15 04:34:56 localhost thinkfan[2635]: This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.
[localhost log]# 
```

maybe this has something to do with those weird last temperatures

vmatare, 2015-12-15:
Thanks again for helping me test this code. f8fb277 should fix one case of memory corruption when reading `/proc/acpi/ibm/thermal`, but I don't know whether that was actually the cause for this segfault. So please test again with the current master :-)

sixsixfive, 2015-12-15:
no sorry still crashes

```
Dec 15 16:00:33 localhost thinkfan[1576]: Using default fan control in /proc/acpi/ibm/fan.
Dec 15 16:00:33 localhost thinkfan[1576]: Using default temperature inputs in /proc/acpi/ibm/thermal.
Dec 15 16:00:33 localhost thinkfan[1576]: sleeptime=5, tmax=51, last_tmax=-128, biased_tmax=51 -> fan="level 7"
Dec 15 16:09:19 localhost thinkfan[1576]: sleeptime=5, tmax=48, last_tmax=47, biased_tmax=-64 -> fan="level 7"
Dec 15 16:09:39 localhost thinkfan[1576]: sleeptime=5, tmax=47, last_tmax=540, biased_tmax=-4003 -> fan="level 7"
Dec 15 16:09:54 localhost thinkfan[1576]: sleeptime=2, tmax=19440, last_tmax=46, biased_tmax=116640 -> fan="
Dec 15 16:09:54 localhost thinkfan[1576]: ERROR: Segmentation fault.
Dec 15 16:09:54 localhost thinkfan[1576]: Backtrace:
Dec 15 16:09:54 localhost thinkfan[1576]: /usr/sbin/thinkfan() [0x447ed5]
/usr/sbin/thinkfan(_ZN8thinkfan5ErrorC2ERKSs+0x5f) [0x448043]
/usr/sbin/thinkfan(_ZN8thinkfan3BugC1ERKSs+0x23) [0x4480e3]
/usr/sbin/thinkfan(_ZN8thinkfan11sig_handlerEi+0xd5) [0x42ddd8]
/usr/sbin/thinkfan(_ZN8thinkfan6LoggerlsERKSs+0x27) [0x440817]
/usr/sbin/thinkfan(_ZN8thinkfan3runERKNS_6ConfigE+0x71d) [0x42e58c]
/usr/sbin/thinkfan(main+0x418) [0x430488]
/usr/sbin/thinkfan(_start+0x29) [0x42da79]
Dec 15 16:09:54 localhost thinkfan[1576]: This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.
[localhost log]# 
```

also my current thinkfan.conf(full speed to hear if it crashes)
```
[hj@localhost ~]$ cat /etc/thinkfan.conf                                                                                                             
(7,0,32767)                                                                                                                                           
[hj@localhost ~]$  
```

vmatare, 2015-12-15:
Ok, at least I can reproduce it now. Looks like some botched math in the biasing logic...

vmatare, 2015-12-17:
"Some botched math" actually turned out to be more like "some botched design", so it took a little longer.
The updated master should now fix this bug, along with a bunch of other issues that had been lingering for some time. However I'm waiting for feedback before I close this since I can't test the specific configuration that triggered it.

sixsixfive, 2015-12-17:
well 

```
/home/hj/thinkfan-git/src/thinkfan-git/src/thinkfan.h:97:25: note:   'thinkfan::temp_state'
 extern TemperatureState temp_state;
                         ^
/home/hj/thinkfan-git/src/thinkfan-git/src/drivers.cpp: At global scope:
/home/hj/thinkfan-git/src/thinkfan-git/src/drivers.cpp:455:1: error: expected declaration before '}' token
 }
 ^
```

vmatare, 2015-12-17:
Sorry, that was a messed up merge. I had to rebase the master, so to be safe it's best if you delete your thinkfan repository and clone it from scratch. Then it should work.

sixsixfive, 2015-12-18:
Ok, it seems it didnt crash overnight so I guess its working now ;)

closed 2015-12-18 by sixsixfive