vmatare/thinkfan #8: Official 1.0 release

sudhirkhanger, 2016-02-24:
Hi,

Do you plan to make an official 1.0 release of thinkfan? I have filed a bug report in Fedora bug tracker so that upstream release can be switched from sourceforge to this Github. That would happen along the line with an official release here. Currently there's 0 releases.

Thanks.

vmatare, 2016-02-25:
Well, I was pondering it after the latest commits. It doesn't seem like I'll get more bug reports by just waiting. But it's going to be called a Beta to alert users to the fact that this branch is very shiny and has seen relatively little testing.

vmatare, 2016-02-25:
Also I really need to improve the documentation before making a release and that puts me off a little ;-)

vmatare, 2016-05-07:
1.0_beta1 has been released.

closed 2016-05-07 by vmatare