vmatare/thinkfan #9: How to activate temperature reading from NVML?

jr-garcia, 2016-03-13:
I need to base the fan speed on the GPU temp. What should the config say to enable NVML? Thanks.

vmatare, 2016-03-20:
Sorry about the delay, was a little busy recently. I'm in the process of writing a new man page for `thinkfan.conf` before making a proper release.

Anyways, for this to work, thinkfan needs to be compiled with `USE_NVML` enabled, i.e. run `cmake` with the option `-DUSE_NVML=ON`. Then you need to configure the PCI bus ID of your nVidia card. For example I get:
`# lspci | grep -i nvidia
01:00.0 VGA compatible controller: NVIDIA Corporation GF108M [NVS 5400M] (rev ff)`

And the config keyword is `nv_thermal`, so I put the following into `/etc/thinkfan.conf`:

`nv_thermal 01:00.0`

Thinkfan should print a short message about the nvidia card if it is opened successfully. And of course you need the proprietary nVidia driver installed, or more specifically, `libnvidia-ml.so` needs to be someplace where libdl can find it.

jr-garcia, 2016-04-07:
Unfortunately my Thinkpad is already suffering for GPU problems so I can not test the solution. But thanks for the answer.

vmatare, 2016-05-07:
There is now a new `thinkfan.conf(5)` manpage that explains this.

closed 2016-05-07 by vmatare