vmatare/thinkfan #10: Problem with fan control for ThinkpadS430

DavidGeisel, 2016-05-03:
Hi! My Lenovo Thinkpad S430 suffers from overheating and i wanted to use thinkfan for it. 
I edited the thinkfan.conf with

`hwmon /sys/devices/virtual/hwmon/hwmon0/temp1_input 
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input 
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input 
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input`
but i get the message

WARNING: Using default fan control in` ``/proc/acpi/ibm/fan.`

`setfan_ibm:` Error writing to /proc/acpi/ibm/fan: Invalid argument
Cleaning up and resetting fan `control.`

The kernel is 4.6.0-040600rc6-generic.

Can you help me with this?

vmatare, 2016-05-03:
What version of thinkfan are you using?

DavidGeisel, 2016-05-03:
thinkfan 0.9.1 and thinkfan 1.0 both didn't work

vmatare, 2016-05-03:
thinkfan 1.0 should give a different message with more information. Did you use the most recent one from github?

vmatare, 2016-05-03:
Please also post your config. To fix this for the 0.9 branch I need to see which fan level format your're using.

DavidGeisel, 2016-05-04:
This is the message received from thinkfan 1.0:

`ERROR:` Unhandled exception: NSt8ios_base7failureB5cxx11E: basic_ios::clear: iostream error.
errno = Permission denied.
Backtrace:./thinkfan() [0x4243fb]
./thinkfan(_ZN8thinkfan15handle_uncaughtEv+0x77) [0x4249c7]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8d6b6) [0x7f6550d316b6]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8d701) [0x7f6550d31701]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8d919) [0x7f6550d31919]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZSt19__throw_ios_failurePKc+0x63) [0x7f6550d5a743]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xf40ca) [0x7f6550d980ca]
./thinkfan(_ZN8thinkfan11TpFanDriverC2ERKSs+0x20e) [0x414d4e]
./thinkfan(_ZN8thinkfan6Config11read_configERKSs+0x6b5) [0x410615]
./thinkfan(main+0x1e9) [0x40a469]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f65506e5830]
./thinkfan(_start+0x29) [0x40ae69]

This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.`

DavidGeisel, 2016-05-04:
[thinkfan_conf.pdf](https://github.com/vmatare/thinkfan/files/248644/thinkfan_conf.pdf)

vmatare, 2016-05-04:
Ah okay, the crucial bit here is `errno = Permission denied` in the 1.0 output. That looks like you're not running thinkfan as root.

If you are indeed running it as root, I need to know your compiler version (output of `gcc -v`). Also, the output does not look like it came from the latest thinkfan version on github. If you cloned the repository, please update it with `git pull` in the main directory. Otherwise just throw away the old thinkfan source code and clone a new copy from github with `git clone https://github.com/vmatare/thinkfan.git`.

madjxatw, 2016-05-07:
I am running thinkfan 0.9.1 on ubuntu 16.04 with kernel 4.4.0-21-generic and only get the warning message: Using default fan control in /proc/acpi/ibm/fan, no more information. Any advice? Need I install version 1.0?

journal of systemd
May 08 01:00:02 *** systemd[1]: Starting simple and lightweight fan control program...
May 08 01:00:02 *** thinkfan[15343]: thinkfan 0.9.1 starting...
May 08 01:00:02 *** thinkfan[15343]: WARNING: Using default fan control in /proc/acpi/ibm/fan.
May 08 01:00:02 *** thinkfan[15343]: Daemon PID: 15345
May 08 01:00:02 *** systemd[1]: Started simple and lightweight fan control program.

vmatare, 2016-06-07:
Commit 44182e8 adds an error message that reminds users to run thinkfan an root. As long as there is no other feedback here I'll have to assume that was the cause of the problem.

closed 2016-09-07 by vmatare

megies, 2017-10-29:
For reference, I just encountered this issue with thinkfan `0.9.2-1` on Debian 8 jessie, installed via official Debian packages. Thinkfan would not start properly after system reboot, resulting in fan not powering up as set in thinkfan config and PC heating more than I would've wished for.

After noticing that PC is heating too much and fan not at full speed, I checked the `thinkfan` service status, finding this:

The fix was trivial, I just restarted the service manually (nothing omitted in between in this paste).

```
$ sudo service thinkfan status
● thinkfan.service - simple and lightweight fan control program
   Loaded: loaded (/lib/systemd/system/thinkfan.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sun 2017-10-29 15:11:52 CET; 6h ago
 Main PID: 2728 (code=exited, status=0/SUCCESS)

Oct 29 15:11:26 wintermute systemd[1]: Starting simple and lightweight fan control program...
Oct 29 15:11:26 wintermute thinkfan[2714]: thinkfan 0.9.1 starting...
Oct 29 15:11:26 wintermute thinkfan[2714]: WARNING: Using default fan control in /proc/acpi/ibm/fan.
Oct 29 15:11:26 wintermute thinkfan[2714]: Daemon PID: 2728
Oct 29 15:11:26 wintermute systemd[1]: Started simple and lightweight fan control program.
Oct 29 15:11:52 wintermute thinkfan[2728]: setfan_ibm: Error writing to /proc/acpi/ibm/fan: Invalid argument
Oct 29 15:11:52 wintermute thinkfan[2728]: Cleaning up and resetting fan control.
$ sudo service thinkfan start
$ sudo service thinkfan status
● thinkfan.service - simple and lightweight fan control program
   Loaded: loaded (/lib/systemd/system/thinkfan.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2017-10-29 21:31:08 CET; 1s ago
  Process: 27775 ExecStart=/usr/sbin/thinkfan (code=exited, status=0/SUCCESS)
 Main PID: 27776 (thinkfan)
    Tasks: 1
   Memory: 268.0K
      CPU: 2ms
   CGroup: /system.slice/thinkfan.service
           └─27776 /usr/sbin/thinkfan

Oct 29 21:31:08 wintermute systemd[1]: Starting simple and lightweight fan control program...
Oct 29 21:31:08 wintermute thinkfan[27775]: Daemon PID: 27776
Oct 29 21:31:08 wintermute systemd[1]: Started simple and lightweight fan control program.
```

I can't really check if issue this is still present in newer packages, as Debian packaging only does security fixes after the release freeze.