vmatare/thinkfan #12: syntax error with complex configuration

GeorgFleig, 2016-06-01:
Hi,
I have some trouble setting up thinkfan using complex mode. I am running the latest version of thinkfan on a T500 which provides 16 sensors. However when using the complex configuration file provided by this repo I get the following error message:

`root@tp:/etc# /usr/local/sbin/thinkfan 

ERROR: /etc/thinkfan.conf:65: Syntax error:
{ level 0   # the fan level
^
`
As far as I understand I don't need to adjust the provided config file as the number of sensors is the same (16). Did I miss something?

Thanks for this great tool and the support!

vmatare, 2016-06-02:
Thanks for the report, this is indeed a bug. Some of the features used in that example config are still missing from the config parser.

EvansMike, 2016-06-08:
Just to say I see a slightly different message using the complex config.

ERROR: Invalid config: Inconsistent limit length

Thanks

vmatare, 2016-06-08:
Yup, there are actually multiple issues here, two of which have been fixed by the two last commits. The third requires some larger changes in the parser which will take some time. As a workaround, you can remove the dots from the fan levels and replace them with sensible temperature limits, or you can use simple mode with correction values.

EvansMike, 2016-06-08:
Thanks, that works. :)

vmatare, 2016-07-23:
This is *supposed* to be fixed by commit #5346b6e, so anyone affected by this might want to retry with the current master.

GeorgFleig, 2016-09-04:
Thanks, seems to be working for me with the current version.

closed 2016-09-07 by vmatare