vmatare/thinkfan #13: Force full-speed mode?

musikerv, 2016-06-06:
Hello everyone.

I have a thinkpad e420.
thinkfan 0.99.0

Is there a way to force the fan to operate at full-speed for some time (or in sothething like on-off mode) for high environment temp conditions????
A simple command?  Or a config script?
Should i use simple or complex config? 
thanks

vmatare, 2016-06-06:
All of these questions should be answered by the config man page
`man thinkfan.conf`
If you haven't read it, please do. If you did and it remains unclear, I'll be happy to improve it if you elaborate on what exactly you're trying to achieve.

musikerv, 2016-06-07:
I opened it and tried but not straightforward how to do it. Something like 'setting level 7 or 255 in /proc/acpi/ibm/fan.
Will try this again later today and report back.
thanks for quick reply...

Andrei-Pozolotin, 2016-06-14:
I confirm that full speed entry works:
``
("level full-speed", 75, 32767) # 5000 rpm
``

musikerv, 2016-06-15:
OK! in Fedora I got 'su' access and then
`echo level full-speed > /proc/acpi/ibm/fan`

then I tested with
`cat /proc/acpi/ibm/fan`
and the output is: 
`status:		enabled
speed:		0
level:		disengaged
commands:	level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:	enable, disable
commands:	watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))
`

so it gets enabled but the level seems to be disengaged...
but with `echo level 7 > /proc/acpi/ibm/fan` it seems to accept the value... 
still cannot hear the fan spinning full, so I am just trying..

p.s. do we still have connection to [these](http://www.thinkwiki.org/wiki/How_to_control_fan_speed) guys? there are some instructions relative to the older thinkfan versions there.

thanks

vmatare, 2016-06-15:
`full-speed` is the same as `disengaged`. You have to wait a minute for the fan speed to ramp up. All of this has nothing to do with thinkfan, you can uninstall it if you just want to control your fan speed manually. The `thinkpad_acpi` kernel module is responsible for `/proc/acpi/ibm/*`. Please read its documentation [here](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/Documentation/laptops/thinkpad-acpi.txt?id=refs/tags/v4.5.7).

closed 2016-06-15 by vmatare

musikerv, 2016-06-15:
OK. I am reading this now. I still use thinkfan, and just in special cases intend to use the manual  feature if at all.
So, in my opinion this topic is actually a documentation related one since I know now how to setup properlly the conf file.

cheers

schplorg, 2024-04-18:
> OK! in Fedora23 I got 'su' access and then: `echo level full-speed > /proc/acpi/ibm/fan`
> 
> then I tested with `cat /proc/acpi/ibm/fan` and the output is: `status: enabled speed: 0 level: disengaged commands: level <level> (<level> is 0-7, auto, disengaged, full-speed) commands: enable, disable commands: watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds)) `
> 
> so it gets enabled but the level seems to be disengaged... but with `echo level 7 > /proc/acpi/ibm/fan` it seems to accept the value... still cannot hear the fan spinning full, so I am just trying..
> 
> p.s. do we still have connection to [these](http://www.thinkwiki.org/wiki/How_to_control_fan_speed) guys? there are some instructions relative to the older thinkfan versions there.
> 
> thanks

This worked great with my Thinkpad T14 Gen3 AMD on Fedora 38