vmatare/thinkfan #15: Implement YAML config (Opinions please!)

vmatare, 2016-06-14:
The config syntax is quite convoluted and maintaining/fixing the parser takes up way too much time. [YAML](https://en.wikipedia.org/wiki/YAML) seems like a good alternative, with an expressive syntax & mature tool support that should reduce maintenance effort considerably.

However, I suppose existing configs will have to be somehow migrated to YAML, so we can't just get rid of the current, messy parser. I don't feel comfortable just dropping support for the current syntax and leaving the problem of migrating configs for the user to deal with.

I'd be grateful for opinions:
- Do we need a conversion tool from old-format to YAML configs?
- Should support for old-format configs be dropped any time in the near future?
- Should any of this happen before 1.0 is officially released?

Andrei-Pozolotin, 2016-06-14:
OPINION=NO:YES:YES

1) Do we need a conversion tool?
* **NO**: *.conf file is small enough

2) Should support for old-format configs be dropped?
* **YES**: drop old-format configs after 1.0

3) Should any of this happen before 1.0 is officially released?
* **YES**: use 1.0 as transition release
* keep old *.conf parser in 1.0
* provide new *.yaml parser in 1.0
* startup logic for 1.0 :  try *.conf first, fall back to *.yaml

dailyherold, 2016-09-02:
Definitely like this idea! Really getting into tweaking my own thinkpad t430s setup as of today, and YAML would be great for the conf file. Especially with the more complex per-sensor setups and additional features added from future development.

vmatare, 2016-09-02:
ok, this is what a YAML config could look like:
```
sensors:
  - hwmon: /sys/devices/platform/coretemp.0
    temps: [1, 3]

  - tp_thermal: /proc/acpi/ibm/thermal

pwm_fan: /sys/class/hwmon....
#tp_fan: /proc/acpi/ibm/fan

levels:
  - speed: 0
    upper_limit: [50, 50, 50]
  
  - speed: 2
    lower_limit: [45, 45, 45]
    upper_limit: [60, 60, 60]
    
  # ...
```
The way the dashes are use does not look very intuitive to me... better ideas?

hungryseashell, 2016-09-06:
Is this why I get this error in 1.0_beta2 (AUR)?

    ERROR: /etc/thinkfan.conf:46: Invalid syntax:
                                       (0,        0,        42)
                                        ^

vmatare, 2016-09-07:
@hungryseashell please report a separate issue about this. Don't forget to mention the exact version you're using (git/beta?) and your distribution. The complete config file would be helpful as well ;-)

sixsixfive, 2016-10-02:
I guess the config file format doesnt matter much but what would make it much easier for the most people is a gui to change the basic options like tpfanco or fancontrol-gui

vmatare, 2016-10-05:
@sixsixfive Originally I deliberately chose not to have a GUI because I wanted to keep things simple from a technical perspective. However I do see that it would help a lot, which is in fact another reason to switch to YAML, because with a GUI we need to **write** the config, too, and YAML simplifies this a lot.

vmatare, 2017-09-08:
YAML config is merged into master. Closing this so new issues are reported individually.

closed 2017-09-08 by vmatare