vmatare/thinkfan #17: No way to handle sensor paths that change?

twhfe, 2016-07-29:
The path to my CPU's sensors sometimes changes on restart. Specifically, the digit after "hwmon" varies. This results in the expected error from thinkfan:

```
/sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input: No such file or directory

/sys/devices/platform/coretemp.0/hwmon/hwmon2/temp4_input: No such file or directory
readconfig: Error getting temperature.
Refusing to run without usable config file!
```
Does thinkfan support any sort of pattern matching in the config file to accommodate this?

vmatare, 2016-07-29:
This problem has existed for some time now with certain distributions. In theory you can work around this by fixing the load order of the relevant kernel modules, however that can be non-trivial depending on your init system.

Solving this properly is surprisingly difficult because there may be multiple instances of a single hwmon driver, all of which have the same name. It might be solvable by allowing hwmon matching either by name, or by PCI ID, so that the user would have to figure out whether an hwmon's name is unique or not...

vmatare, 2016-11-01:
I've begun implementing a solution for this problem along with the new YAML config format. It supports specifying only a base path for any given hwmon driver, along with a number of indices that specify which `temp*_input` files should be used from that hwmon. Thinkfan then recursively searches the base path for the appropriate `temp*_input` files. Anyone interested in this can check out the `yaml-config` branch.

vmatare, 2017-04-20:
This feature is now merged into the current master branch. The YAML config is still missing some documentation, but at least there's an example config that shows how to use sensor/fan paths with indices, which makes the config robust against changes in driver load order.

vmatare, 2017-09-08:
Closing since the new feature is merged. Continuing with the new related issues that came up.

bssb, 2019-02-04:
Sorry to comment on a closed issue, but I am having the same issue as described by @lynxite and the fix you mentioned seems to address a different problem.

> specifying only a base path for any given hwmon driver, along with a number of indices that specify which temp*_input files should be used from that hwmon

It is not the temp*_input digit that changes, it is the hwmon* digit. Is there any way to handle when the hwmon base path changes between hwmon2 and hwmon3?

vmatare, 2019-02-04:
> It is not the temp*_input digit that changes, it is the hwmon* digit. Is there any way to handle when the hwmon base path changes between hwmon2 and hwmon3?
That's exactly what this feature does. You specify the `temp*_input` indices precisely because they do not change, but leave out the `hwmon*` index because it *does* change.
So let's say I have:
`/sys/devices/platform/nct6775.2592/hwmon/hwmon2`
And the 2 may change. Then I leave the last component out:
```
sensors:
   - hwmon: /sys/devices/platform/nct6775.2592/hwmon/hwmon2
     indices: [1, 3]
```
And specify indices 1 and 3 to use `temp1_input` and `temp3_input`, regardless of the `hwmon*` index. See `examples/thinkfan.conf.yaml` for a full example.

bssb, 2019-02-04:
Ok I have it working now. I had read the example config but it wasn't clear to me that in
```
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [2,3,4,5]
```
actually refers to the path ```/sys/devices/platform/coretemp.0/hwmon/hwmon*/```. Thank you for clarifying.

NeoFromMatrix, 2019-10-31:
Thanks a lot for implementing the feature, I was stuck with the same situation. 

Could you maybe add documentation directly to the config file? How it operates, how the paths in the config file correspond to the paths in the file system and how to set it up. 

It was hard to understand that it will automatically look for temp[indices]_input in all sub directories of a given base path as this behavior is quite different compared to the other config files.

desdic, 2020-01-02:
Maybe I'm not doing it right but I'm having problems with thinkfan finding the temperatures. I have a thinkfan.conf that works but I would really like to use the yaml options since my hardware monitors change on every boot. So I have this thinkfan.yaml

sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [3, 4, 1, 5, 2]

fans:
  - tp_fan /proc/acpi/ibm/fan

levels:
  - speed: 0
    lower_limit: [0, 0, 0]
    upper_limit: [59, 59, 59]

  - speed: 1
    lower_limit: [52, 52, 52]
    upper_limit: [75, 75, 75]

  - speed: 2
    lower_limit: [65, 65, 65]
    upper_limit: [78, 78, 78]

  - speed: 3
    lower_limit: [66, 66, 66]
    upper_limit: [76, 76, 76]

  - speed: 4
    lower_limit: [67, 67, 67]
    upper_limit: [82, 82, 82]

  - speed: 5
    lower_limit: [68, 68, 68]
    upper_limit: [85, 85, 85]

  - speed: 7
    lower_limit: [75, 75, 75]
    upper_limit: [32767, 32767, 32767]

and the error I get is

thinkfan -c ./thinkfan.yaml

ERROR: Unhandled YAML::BadSubscript: operator[] call on a scalar (key: "hwmon").
errno = No such file or directory.
Backtrace:
thinkfan(+0x37654) [0x561ec9680654]
thinkfan(+0x2173b) [0x561ec966a73b]
/usr/lib/libstdc++.so.6(+0xa34da) [0x7fcee979d4da]
/usr/lib/libstdc++.so.6(+0xa3537) [0x7fcee979d537]
/usr/lib/libstdc++.so.6(+0xa378e) [0x7fcee979d78e]
thinkfan(YAML::Node const YAML::Node::operator[]<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const+0x63e) [0x561ec968e3ee]
thinkfan(YAML::convert<std::vector<YAML::wtf_ptr<thinkfan::FanDriver>, std::allocator<YAML::wtf_ptr<thinkfan::FanDriver> > > >::decode(YAML::Node const&, std::vector<YAML::wtf_ptr<thinkfan::FanDriver>, std::allocator<YAML::wtf_ptr<thinkfan::FanDriver> > >&)+0x1ad) [0x561ec968fd4d]
thinkfan(YAML::convert<YAML::wtf_ptr<thinkfan::Config> >::decode(YAML::Node const&, YAML::wtf_ptr<thinkfan::Config>&)+0x14c2) [0x561ec9684532]
thinkfan(thinkfan::Config::try_read_config(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x254) [0x561ec9670d44]
thinkfan(thinkfan::Config::read_config(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x1a) [0x561ec967193a]
thinkfan(main+0x20e) [0x561ec966b8ae]
/usr/lib/libc.so.6(__libc_start_main+0xf3) [0x7fcee9540153]
thinkfan(_start+0x2e) [0x561ec966c3ae]

This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.
Aborted (core dumped)

version 1.0.2.r16.7027a2d-1

vmatare, 2020-04-04:
Created issue #87 to keep track of missing YAML config documentation, because this is a technical issue that should be fixed in the current master and in the 1.1 release.

closed 2020-04-04 by vmatare

thobianchi, 2020-11-29:
> actually refers to the path `/sys/devices/platform/coretemp.0/hwmon/hwmon*/`. Thank you for clarifying.

This is not exact @bssb . Like vmatare said:
> And specify indices 1 and 3 to use temp1_input and temp3_input, regardless of the hwmon* index.

from the man page `man thinkfan.conf 5`

> 2) A directory that contains a specific hwmon driver, for example “/sys/devices/platform/nct6775.2592”. 
> Note that this path does not contain the load-order dependent “hwmonX” folder.
> As long as it contains only a single hwmon driver/interface it is sufficient to specify the “indices: index-list” entry to tell
> thinkfan which specific sensors to use from that interface. The “name: hwmon-name” entry is unnecessary.

for `specific sensor to use` it refers to `temp<index>_input` file

vmatare, 2020-11-29:
@thobianchi Note that the config manpage has been updated several times now, so people that read one of the older versions may well have been confused by some unclear or incomplete descriptions that used to be in there. Also keep in mind that getting feedback from people who misunderstood something is instrumental to improving the quality of documentation ;-)

Reutertu3, 2021-03-07:
After figuring out that thinkfan.conf isn't exactly the same as thinkfan.yaml, yet both can be used interchangeably (every documentary on the web other than the official really is outdated), I was still running into the problem that the sensors on my Laptop couldn't be found upon boot.

`thinkfan[323]: ERROR: /etc/thinkfan.yaml:8:
                name: coretemp
                                    ^
                Could not find a hwmon with this name.
audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=thinkfan comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
systemd[1]: thinkfan.service: Control process exited, code=exited, status=1/FAILURE
systemd[1]: thinkfan.service: Failed with result 'exit-code'.
systemd[1]: Failed to start simple and lightweight fan control program.
`

Thinkfan and the config work flawlessly after startup. I solved it by delaying the systemd unit with _After=multi-user.target_