vmatare/thinkfan #20: unhandle exceptio

atussec, 2016-09-01:
install beta2 but getting this error

```
> ERROR: Unhandled exception: NSt8ios_base7failureB5cxx11E: basic_ios::clear: iostream error.
> errno = No such file or directory.
> Backtrace:./thinkfan() [0x420d1a]
> ./thinkfan(_ZN8thinkfan15handle_uncaughtEv+0x67) [0x4210e7]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8f056) [0x7f4cf146b056]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8f0a1) [0x7f4cf146b0a1]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8f2b9) [0x7f4cf146b2b9]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZSt19__throw_ios_failurePKc+0x53) [0x7f4cf14939c3]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xf33ad) [0x7f4cf14cf3ad]
> ./thinkfan(_ZN8thinkfan12SensorDriverC2ESs+0x16d) [0x41460d]
> ./thinkfan(_ZN8thinkfan14TpSensorDriverC2ESs+0x2b) [0x415eab]
> ./thinkfan(_ZNK8thinkfan12SensorParser6_parseERPKc+0x1aa) [0x41ae3a]
> ./thinkfan(_ZNK8thinkfan12ConfigParser6_parseERPKc+0x470) [0x420060]
> ./thinkfan(_ZN8thinkfan6Config11read_configERKSs+0x202) [0x40f222]
> ./thinkfan(main+0x1bb) [0x409bbb]
> /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f4cf0b40700]
> ./thinkfan(_start+0x29) [0x40a519]
```

vmatare, 2016-09-02:
Ah sorry, that's not a good error message. One of the files from your sensor setup seems to be missing (double-checked the paths?)

vmatare, 2016-11-01:
Closing because this type of error should generate a much more helpful error message in the current master now.

closed 2016-11-01 by vmatare