vmatare/thinkfan #21: [systemd] thinkfan starts too early?

sixsixfive, 2016-10-02:
thinkfan is unable to autanatically run on my system now however if i start it after the pc has fully booted up it works so i guess it depends on anything that isn't there at boot

```
[ssf@hj-pc ~]$ systemctl status thinkfan
● thinkfan.service - Thinkfan, the minimalist fan control program
   Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since So 2016-10-02 22:34:24 CEST; 1min 19s ago
  Process: 268 ExecStart=/usr/bin/thinkfan (code=exited, status=1/FAILURE)

Okt 02 22:34:24 hj-pc systemd[1]: Starting Thinkfan, the minimalist fan control program...
Okt 02 22:34:24 hj-pc systemd[1]: thinkfan.service: Control process exited, code=exited status=1
Okt 02 22:34:24 hj-pc systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
Okt 02 22:34:24 hj-pc systemd[1]: thinkfan.service: Unit entered failed state.
Okt 02 22:34:24 hj-pc systemd[1]: thinkfan.service: Failed with result 'exit-code'.
[ssf@hj-pc ~]$ 
```

```
[ssf@hj-pc ~]$ systemd-analyze blame    
           682ms dev-sda1.device
           433ms systemd-backlight@backlight:intel_backlight.service
           362ms upower.service
           142ms systemd-vconsole-setup.service
           120ms systemd-journald.service
           119ms systemd-udevd.service
           104ms systemd-rfkill.service
           100ms systemd-udev-trigger.service
            88ms polkit.service
            73ms NetworkManager.service
            51ms systemd-tmpfiles-setup.service
            43ms udisks2.service
            41ms systemd-logind.service
            40ms user@1000.service
            38ms kmod-static-nodes.service
            37ms dev-mqueue.mount
            35ms systemd-modules-load.service
            30ms systemd-sysctl.service
            30ms sys-kernel-config.mount
            29ms dev-hugepages.mount
            25ms sys-kernel-debug.mount
            24ms systemd-remount-fs.service
            21ms systemd-random-seed.service
            21ms systemd-tmpfiles-setup-dev.service
            19ms ntpd.service
            17ms avahi-daemon.service
            16ms thinkfan.service
            15ms systemd-journal-flush.service
            14ms bluetooth.service
            11ms wpa_supplicant.service
             9ms systemd-update-utmp.service
             6ms rtkit-daemon.service
             6ms systemd-user-sessions.service
             5ms tmp.mount
lines 3-34/34 (END)
```

sixsixfive, 2016-10-03:
sorry but i guess this was an issue with a broken systemd on my system

closed 2016-10-03 by sixsixfive

cmosig, 2019-12-13:
> sorry but i guess this was an issue with a broken systemd on my system

I am having the same problem. How did you solve it? What was broken in systemd?