vmatare/thinkfan #23: openrc displays thinkfan as crashed after latest commit

free-city-of-ulm, 2016-11-01:
` thinkfan  [  crashed  ]`
It also does not seem to start at all as it does not show up in ps aux.

vmatare, 2016-11-01:
Thanks for the report. Can you post your `/etc/init.d/thinkfan` file as well as the output of `make install` after compiling thinkfan?

vmatare, 2016-11-01:
It might also help to see the entire output of you trying start the init script, i.e. what happens when you run `/etc/init.d/thinkfan start` or whatever is correct for your distro.

free-city-of-ulm, 2016-11-01:
make install
```
[100%] Built target thinkfan
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/sbin/thinkfan
-- Up-to-date: /usr/local/share/doc/thinkfan/COPYING
-- Up-to-date: /usr/local/share/doc/thinkfan/README
-- Up-to-date: /usr/local/share/doc/thinkfan/thinkfan.conf.complex
-- Up-to-date: /usr/local/share/doc/thinkfan/thinkfan.conf.simple
-- Up-to-date: /usr/local/share/man/man1/thinkfan.1
-- Up-to-date: /usr/local/share/man/man5/thinkfan.conf.5
-- Up-to-date: /etc/init.d/thinkfan
```

sudo /etc/init.d/thinkfan start
```
thinkfan         | * /etc/init.d/thinkfan uses runscript, please convert to openrc-run.
thinkfan         | * Starting thinkfan ...
thinkfan         |
thinkfan         |Daemon PID: 5624                           [ ok ]
```

[thinkfan.txt](https://github.com/vmatare/thinkfan/files/564967/thinkfan.txt)

vmatare, 2016-11-01:
Looks good so far, now is there a process with PID 5624? If not there should be some kind of error message in syslog...

free-city-of-ulm, 2016-11-01:
Sorry, I probably should have posted this first.  This is from syslog-ng.
```
Nov  1 16:47:01 ulm sudo[10765]:    chris : TTY=pts/1 ; PWD=/home/chris/Documents/thinkfan/build ; USER=root ; COMMAND=/sbin/rc-service thinkfan start
Nov  1 16:47:01 ulm /etc/init.d/thinkfan[10765]: /etc/init.d/thinkfan uses runscript, please convert to openrc-run.
Nov  1 16:47:01 ulm thinkfan[10785]: ERROR: Unhandled std::ios_base::failure: basic_ios::clear.
Nov  1 16:47:01 ulm thinkfan[10785]: errno = No such file or directory.
Nov  1 16:47:01 ulm thinkfan[10785]: _GLIBCXX_USE_CXX11_ABI = 1.
Nov  1 16:47:01 ulm thinkfan[10785]: Backtrace:
Nov  1 16:47:01 ulm thinkfan[10785]: /usr/local/sbin/thinkfan() [0x45c70d]
/usr/local/sbin/thinkfan(thinkfan::handle_uncaught()+0xef) [0x45cfbd]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.2.0/libstdc++.so.6(+0xb4265) [0x7f53fa47e265]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.2.0/libstdc++.so.6(+0xb4350) [0x7f53fa47e350]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.2.0/libstdc++.so.6(+0xb46fb) [0x7f53fa47e6fb]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.2.0/libstdc++.so.6(std::__throw_ios_failure(char const*)+0x7f) [0x7f53fa4b1f8f]
/usr/lib/gcc/x86_64-pc-linux-gnu/6.2.0/libstdc++.so.6(std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)+0x61) [0x7f53fa5054b1]
/usr/local/sbin/thinkfan(thinkfan::PidFileHolder::PidFileHolder(int)+0xbc) [0x4307fe]
/usr/local/sbin/thinkfan(main+0x3c1) [0x4314b6]
/lib64/libc.so.6(__libc_start_main+0xf1) [0x7f53f9b36651]
/usr/local/sbin/thinkfan(_start+0x29) [0x42eb79]
Nov  1 16:47:01 ulm thinkfan[10785]: This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.
```

vmatare, 2016-11-01:
OK, another weird CMake fuckup. Should be fixed by 8636362 in the current master.

free-city-of-ulm, 2016-11-01:
Fixed, thanks

closed 2016-11-01 by free-city-of-ulm