vmatare/thinkfan #24: Systemd issue?

TullyMore, 2016-11-02:
Hi,

I'm using thinkfan for a couple of years now on my Lenovo T420s without any issues. Thanks for the great program.

Lately I upgraded to Ubuntu 16.10 and found that the fan was constantly running using the thinkfan version provided by Ubuntu. 

So I decided to download 'Thinkfan version: 1.0_beta2' and built it using
`cmake -D CMAKE_BUILD_TYPE:STRING=Debug -D USE_NVML:BOOL=ON .`

Unfortunately the behaviour stayed the same.

When started with `systemctl start thinkfan` the output of 'sensors' is:

acpitz-virtual-0
Adapter: Virtual device
temp1:        +47.0°C  (crit = +97.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +53.0°C  (high = +86.0°C, crit = +100.0°C)
Core 0:         +53.0°C  (high = +86.0°C, crit = +100.0°C)
Core 1:         +50.0°C  (high = +86.0°C, crit = +100.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        3997 RPM

When idle the fan speed usually sicks around 4000 RPM. But as you can see from my '/etc/thinkfan.conf' file at the bottom of this post, the fan should only start at temperatures above 55 degrees.

The output of `systemctl status thinkfan` is:

● thinkfan.service - LSB: thinkfan initscript
   Loaded: loaded (/etc/init.d/thinkfan; bad; vendor preset: enabled)
   Active: active (exited) since Mi 2016-11-02 11:06:58 AEDT; 47min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 5941 ExecStop=/etc/init.d/thinkfan stop (code=exited, status=0/SUCCESS)
  Process: 6708 ExecStart=/etc/init.d/thinkfan start (code=exited, status=0/SUCCESS)

Nov 02 11:06:58 moon systemd[1]: Starting LSB: thinkfan initscript...
Nov 02 11:06:58 moon systemd[1]: Started LSB: thinkfan initscript.
Nov 02 11:49:57 moon systemd[1]: Started LSB: thinkfan initscript.

If I now stop thinkfan using `systemctl stop thinkfan` and start it using `thinkfan -n -b 1 -s 3 -v -v -v  -c /etc/thinkfan.conf` you can see the fan to immediately start using the speed levels as defined in '/etc/thinkfan.conf' and thinkfan is working as expected.

# thinkfan -n -b 1 -s 3 -v -v -v -c /etc/thinkfan.conf 
Temperatures(bias): 52(0), 51(0), 48(0), 46(0) -> level 0
Temperatures(bias): 52(0), 51(0), 48(0), 46(0)
Temperatures(bias): 54(0), 56(1), 51(0), 47(0) -> level 1
Temperatures(bias): 50(0), 50(0), 50(0), 48(0)

The systemd service seems to be starting and running correctly, but doesn't seem to be using the settings in '/etc/thinkfan.conf'. Any idea how to solve this issue would be appreciated.

System, settings and config files:

Kernel: 4.4.0-43-generic

Graphic card: Nvidia Optimus NVS 4200M (switchable)
GPU driver: NVIDIA binary driver - version 370.28 from nvidia-370 (open source)

`cat /etc/default/thinkfan`

# Should thinkfan be started automatically on boot?
# Only say "yes" when you know what you are doing, have configured
# thinkfan correctly for *YOUR* machine and loaded thinkpad_acpi
# with fan_control=1 (if you have a ThinkPad).
START=yes

# Additional startup parameters
#DAEMON_ARGS="-q"
DAEMON_ARGS="-q -b 1 -s 3"

`cat /etc/modprobe.d/thinkfan.conf`

options thinkpad_acpi fan_control=1

`cat /etc/thinkfan.conf`

######################################################################
# thinkfan 0.7 example config file
# ================================
#
# ATTENTION: There is only very basic sanity checking on the configuration.
# That means you can set your temperature limits as insane as you like. You
# can do anything stupid, e.g. turn off your fan when your CPU reaches 70°C.
#
# That's why this program is called THINKfan: You gotta think for yourself.
#
######################################################################
#
# IBM/Lenovo Thinkpads (thinkpad_acpi, /proc/acpi/ibm)
# ====================================================
#
# IMPORTANT:
#
# To keep your HD from overheating, you have to specify a correction value for
# the sensor that has the HD's temperature. You need to do this because
# thinkfan uses only the highest temperature it can find in the system, and
# that'll most likely never be your HD, as most HDs are already out of spec
# when they reach 55 °C.
# Correction values are applied from left to right in the same order as the
# temperatures are read from the file.
#
# For example:
# tp_thermal /proc/acpi/ibm/thermal (0, 0, 10)
# will add a fixed value of 10 °C the 3rd value read from that file. Check out
# http://www.thinkwiki.org/wiki/Thermal_Sensors to find out how much you may
# want to add to certain temperatures.

#  Syntax:
#  (LEVEL, LOW, HIGH)
#  LEVEL is the fan level to use (0-7 with thinkpad_acpi)
#  LOW is the temperature at which to step down to the previous level
#  HIGH is the temperature at which to step up to the next level
#  All numbers are integers.
#

# I use this on my T61p:
tp_fan /proc/acpi/ibm/fan
#tp_thermal /proc/acpi/ibm/thermal (0, 10, 15, 2, 10, 5, 0, 3, 0, 3)

# Find sensors using: find /sys/devices -type f -name "temp*_input"
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
hwmon /sys/devices/virtual/hwmon/hwmon0/temp1_input

# Currently deactivated
#nv_thermal 01:00.0

(0, 0, 55)
(1, 48, 60)
(2, 50, 61)
(3, 52, 63)
(4, 56, 65)
(5, 59, 67)
(7, 63, 32767)

`cat /proc/acpi/ibm/fan`
status:		enabled
speed:		3520
level:		3
commands:	level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:	enable, disable
commands:	watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))

vmatare, 2016-11-02:
Well, the contents of `/proc/acpi/ibm/fan` look like thinkfan has (correctly?) set the fan level to 3. What do you see in `/proc/acpi/ibm/fan` when you feel like it's not working? If you suspect it's using the wrong config file, just look at the commandline args of the process, i.e. do `ps ax | grep thinkfan`. Also, you might want to ditch the beta2 (it has some rather serious bugs) and try the most recent version: `git clone https://github.com/vmatare/thinkfan.git`

TullyMore, 2016-11-02:
Hello Victor. Thanks for your lightning quick reply. I downloaded and installed the most recent version and everything works like a charm now. Guess I probably ran in one of the mentioned bugs.

As in above mentioned case the output of `sensors` was showing a temperature of 53 degrees. With my configuration I did expect the fan to be set to level 1 instead of 3. This happened correctly when thinkfan was started on the cli.

Regarding the output of `ps aux | grep thinkfan`. With the buggy version I didn't get any output. I'm no expert with systemd, but I assume from the output of `systemctl status thinkfan` stating `/etc/init.d/thinkfan; bad; vendor preset: enabled)`, that thinkfan failed to load and the presets were enabled, thus no result with `ps`. `systemctl status thinkfan` output now looks like this:

\# systemctl status thinkfan
● thinkfan.service - Thinkfan, the minimalist fan control program
   Loaded: loaded (/usr/local/lib/systemd/system/thinkfan.service; enabled; vend
  Drop-In: /etc/systemd/system/thinkfan.service.d
           └─override.conf
   Active: active (running) since Mi 2016-11-02 14:01:35 AEDT; 7min ago
  Process: 24575 ExecStart=/usr/local/sbin/thinkfan $THINKFAN_ARGS (code=exited,
 Main PID: 24577 (thinkfan)
   CGroup: /system.slice/thinkfan.service
           └─24577 /usr/local/sbin/thinkfan -b0

With the current version running `ps aux | grep thinkfan` returns `24577 ?        S      0:00 /usr/local/sbin/thinkfan -b0`. Shouldn't I expect seeing the arguments set in `/etc/default/thinkfan` which in my case are `DAEMON_ARGS="-q -b 1 -s 3"` instead of `-b0`?

Either way. thinkfan seems to be running smoothly now. Thanks again for the program and your help.

vmatare, 2016-11-02:
Right, on Ubuntu we're supposed to use the arguments specified in `/etc/default/thinkfan`. However I think that file stems from the old init system. With systemd there is another default location for service configs, as shown by your `systemctl status thinkfan`:
```
[...]
Drop-In: /etc/systemd/system/thinkfan.service.d
└─override.conf
```
The nice thing about this file is that you don't even have to know about it, you can just do `systemctl edit thinkfan` to change the commandline arguments.

vmatare, 2016-11-02:
FWIW, recent versions of thinkfan should not install `/etc/default/thinkfan` any more, at least not when built & installed with CMake. So that file should in fact be a leftover from your previous thinkfan installation. However I don't know the current policy on Ubuntu, i.e. whether they somehow adapt systemd and want to keep the old `/etc/default/*` system around. One way or another, I'd leave that for the package maintainers to sort out, so I guess you should delete `/etc/default/thinkfan` and go with the `systemctl edit thinkfan` way, at least until there's a new .deb released for Ubuntu.

TullyMore, 2016-11-03:
You are right. Inserting `Environment='THINKFAN_ARGS=-q -b 2 -s 3'` into the override.conf file with `systemctl edit thinkfan` as suggested works as expected. As a fallback I'll keep my `/etc/default/thinkfan` with a comment how to properly set these settings.

Is there any documentation where I could have found this information? I have been following various howtos found on the net and have the feeling that many of them are a little outdated. I understand that it's not easy to keep up with all the system changes and various distros, but I would be happy to contribute our latest findings in a more prominent spot than this issue thread.

TullyMore, 2016-11-03:
Maybe you could help me with a different issue as well. I encounter the problem that with level 7 the fan isn't using it's full potential. I found your workaround here [https://sourceforge.net/p/thinkfan/thinkfan-bugs/10/](url) and started thinkfan in dangerous mode (-D) with various values and combinations replacing 7 in the level 7 line:
`"level full-speed"`, `level full-speed`, `"full-speed"`, `full-speed`
`"level disengaged"`, `level disengaged`, `"disengaged"`, `disengaged`

Each combination resulted in an 'Invalid syntax' error message by thinkfan. Interestingly enough though, the fan started to run at maximum rpm as tested with `echo "level full-speed" > /proc/acpi/ibm/fan` and didn't slow down, despite the temperature was way below threshold. Unnecessary to say that starting the `systemd` service failed. Any suggestion?

vmatare, 2016-11-03:
Hey, for future reference, when you come up with a new issue, please open a separate github issue about it. I know it's not intuitive because the conversation is right here, but we want to make it as easy as possible for others to find answers if they have the same question.

That said, I can't reproduce the problem you're having with `"level full-speed"`. Have you tested this with the current version? Early versions of the 1.0 branch had a bug where `"level full-speed"` wasn't recognized, but it's fixed now and I'm using it myself. If the problem indeed persists with the current version, please open a **new** issue about it and post your entire config file there.

TullyMore, 2016-11-04:
Thanks. I will close this issue and open a new one.

closed 2016-11-04 by TullyMore