vmatare/thinkfan #26: "level full-speed" rejected as invalid syntax

TullyMore, 2016-11-04:
I encounter the problem that with level 7 the fan isn't using it's full potential. I found your workaround here https://sourceforge.net/p/thinkfan/thinkfan-bugs/10/ and started thinkfan in dangerous mode (-D) with various values and combinations replacing 7 in the level 7 line:
"level full-speed", level full-speed, "full-speed", full-speed
"level disengaged", level disengaged, "disengaged", disengaged

Each combination resulted in an "Invalid syntax" error message by thinkfan Version 0.99.0 cloned from https://sourceforge.net/projects/thinkfan/ on 4. Nov. 2016. 

` \# thinkfan -n -D `
`ERROR: /etc/thinkfan.conf:57: Invalid syntax:`
`("level full-speed", 68, 32767)`

~~Interestingly enough though, the fan started to run at maximum rpm as tested with echo "level full-speed" > /proc/acpi/ibm/fan and didn't slow down, despite the temperature was way below threshold. Unnecessary to say that starting the systemd service failed.~~ (This behaviour occurred with an older build)

`
\# That's why this program is called THINKfan: You gotta think for yourself.
\#
\######################################################################
\#
\# IBM/Lenovo Thinkpads (thinkpad_acpi, /proc/acpi/ibm)
\# ====================================================
\#
\# IMPORTANT:
\#
\# To keep your HD from overheating, you have to specify a correction value for
\# the sensor that has the HD's temperature. You need to do this because
\# thinkfan uses only the highest temperature it can find in the system, and
\# that'll most likely never be your HD, as most HDs are already out of spec
\# when they reach 55 °C.
\# Correction values are applied from left to right in the same order as the
\# temperatures are read from the file.
\#
\# For example:
\# tp_thermal /proc/acpi/ibm/thermal (0, 0, 10)
\# will add a fixed value of 10 °C the 3rd value read from that file. Check out
\# http://www.thinkwiki.org/wiki/Thermal_Sensors to find out how much you may
\# want to add to certain temperatures.

\#  Syntax:
\#  (LEVEL, LOW, HIGH)
\#  LEVEL is the fan level to use (0-7 with thinkpad_acpi)
\#  LOW is the temperature at which to step down to the previous level
\#  HIGH is the temperature at which to step up to the next level
\#  All numbers are integers.
\#

\# I use this on my T61p:
tp_fan /proc/acpi/ibm/fan
\#tp_thermal /proc/acpi/ibm/thermal (0, 10, 15, 2, 10, 5, 0, 3, 0, 3)

hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
hwmon /sys/devices/virtual/hwmon/hwmon0/temp1_input

\#nv_thermal 01:00.0

(0, 0, 52)
(1, 46, 59)
(2, 54, 65)
(3, 58, 69)
(4, 62, 72)
(5, 65, 74)
("level full-speed", 68, 32767)
\#(7, 68, 32767)
`

vmatare, 2016-11-06:
Thanks for taking the time to report this. There was indeed a rather specific bug in the parser, which should be fixed by e806611. Please go ahead and close this issue if you find that the problem is solved.

TullyMore, 2016-11-07:
It works now. Thanks for the fix.

Hint: Make sure to add/re-add the `-D` option using `systemctl edit thinkfan` to use the `"level full-speed"` option as the new install will overwrite any previously set DAEMON_ARGS.

closed 2016-11-07 by TullyMore

vmatare, 2016-11-07:
Oh right, the install will overwrite any user-defined settings in `override.conf`. That's in fact issue #27 ;-)

stobenski, 2017-11-04:
```"level full-speed"```

we need this in docs or config examples.