vmatare/thinkfan #28: Sytanx Error With Simple Configuration

JesusIslam, 2016-11-30:
This is my `/etc/thinkfan.conf`:

```
hwmon /sys/devices/virtual/hwmon/hwmon0/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input

(0,    0,    55)
(1,    48,    60)
(2,    50,    61)
(3,    52,    63)
(4,    56,    65)
(5,    59,    66)
(7,    63,    32767)
```

When I do `sudo thinkfan -n` it says 
```
ERROR: /etc/thinkfan.conf:6: Invalid syntax:
(0,    0,    55)
```

Am I doing something wrong?

vmatare, 2016-12-01:
Yup, another bug in the parser. As a workaround, put some comment before the fan levels, i.e. like:

```
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input

# Useless comment line to work around parser bug

(0,    0,    55)
```

vmatare, 2016-12-01:
To be honest, this is really not just a parser bug, but rather a design fault in the config syntax. High time I finished the YAML config I guess. Anyways, thanks for reporting, bugfix is coming up...

vmatare, 2016-12-01:
If you want to help, please clone the current `master` from github and try it out. Should be fixed there. Make sure to remove the workaround-comment in case you did put that in your config.

JesusIslam, 2016-12-01:
@vmatare after I pull the latest commit, it works perfectly now. I guess yaml support would be great.

closed 2016-12-01 by JesusIslam