vmatare/thinkfan #29: biased_tmax seems very high

tom-mayer, 2016-12-04:
Hey,

first up, been using thikfan for years and its really great! I just installed a clean version of Arch (been using it on Xubuntu before) and never had this issue. This is the output from the deamon (-n):

```
sleeptime=5, tmax=48, last_tmax=48, biased_tmax=48 -> fan="level 1"
sleeptime=5, tmax=45, last_tmax=46, biased_tmax=45 -> fan="level 0"
sleeptime=2, tmax=55, last_tmax=46, biased_tmax=68 -> fan="level 1"
sleeptime=5, tmax=45, last_tmax=47, biased_tmax=45 -> fan="level 0"
```

The line where it swaps to level 1 is the interesting one:
```
sleeptime=2, tmax=55, last_tmax=46, biased_tmax=68 -> fan="level 1"
```

I'm not 100% sure what biased_tmax is so I'm taking a wild guess and its the projected temperature that will be reached in a certain time interval when the temperature keeps rising at the current rate (last_tmax -> tmax is pretty steep here).

The fan configuration says it should never go to lvl1 before 65.  When I watch my real sensor values it never goes above 56 or 57, even with the steep rise. The fan comes up regardless since it seems to take the biased_tmax.

Is there any way to maybe update faster or control this? I've seen that there is a -b option for the daemon which you can put in DAEMON_ARGS in /etc/default/thinkfan, but it doesn't seem to affect the behavior. Arch is systemd based, does that file even get used there? (my first systemd system, so sry if this is a noob mistake :))

Cheers

vmatare, 2016-12-05:
First of all, if you start thinkfan on the command line, the `-b` option should definitely affect behavior. E.g. with `-b0`, `biased_tmax` should always be the same as `tmax`. If you're not sure whether `/etc/default/thinkfan` is actually used, check the command line of the thinkfan process:
`# ps fax | grep thinkfan`.
It looks like you're using one of the 0.9 releases that was probably shipped packaged with your distribution, so if there's any problem `/etc/default/thinkfan`, you'll have to complain to your distro's package maintainer about that.

vmatare, 2016-12-05:
On a more general note, the default bias for the `-b` option is `15`, which is pretty high. In that third line, temperature jumps up by 9 °C, so thinkfan `biased_tmax` = `tmax` + 9 * BIAS/10 = 9 * 1.5 = 13 °C to the real temperature. If you have e.g. an Intel i-series processor, that's not a good setting. What you rather want is a **negative** bias (e.g. `-b-5`) to counter the processor's jumpy temperature response, so that in line 3 you'd get `biased_tmax` = 55 + 9 * -0.5 = 51 °C.

vmatare, 2016-12-05:
You could also post your `/usr/lib/systemd/system/thinkfan.service`, or wherever `systemctl status thinkfan` says you service file is.

tom-mayer, 2016-12-05:
Hey,

thanks a lot for the pointers, especially the one with the negative bias, would have never thought to use it that way. It turned out that I had to override the the unit file with systemctl edit thinkfan and override the ExecStart and append the arguemtns there. Works perfectly now as before. Turned out not to be a thinkfan issue at all, but I'm glad I understood bias now and can use it to smooth out the spikes in my i7 when it's only clocking up for a short period of time, now I dont get fan stuttering at all.

closed 2016-12-05 by tom-mayer