vmatare/thinkfan #32: Cannot seem to start thinkfan on system startup (Ubuntu 16.04, version 0.91)

arooni, 2017-01-21:
Followed http://askubuntu.com/questions/790576/thinkfan-does-not-start-automatically as a guide, namely ran sudo systemctl enable thinkfan.service ; and sudo systemctl enable thinkfan

But now when I try to run 
[I]  ✘  /  sudo /etc/init.d/thinkfan start
/etc/init.d/thinkfan: 10: /etc/default/thinkfan: DAEMON_ARGS: not found
Starting thinkfan (via systemctl): thinkfan.serviceJob for thinkfan.service failed because the control process exited with error code. See "systemctl status thinkfan.service" and "journalctl -xe" for details.
 failed!

[I]  ✘  /  systemctl status thinkfan.service
● thinkfan.service - simple and lightweight fan control program
   Loaded: loaded (/lib/systemd/system/thinkfan.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-01-20 21:52:53 CST; 19s ago
  Process: 25038 ExecStart=/usr/sbin/thinkfan (code=exited, status=4)

Jan 20 21:52:53 LilArooni systemd[1]: Starting simple and lightweight fan control program...
Jan 20 21:52:53 LilArooni thinkfan[25038]: thinkfan 0.9.1 starting...
Jan 20 21:52:53 LilArooni systemd[1]: thinkfan.service: Control process exited, code=exited status=4
Jan 20 21:52:53 LilArooni systemd[1]: Failed to start simple and lightweight fan control program.
Jan 20 21:52:53 LilArooni systemd[1]: thinkfan.service: Unit entered failed state.
Jan 20 21:52:53 LilArooni systemd[1]: thinkfan.service: Failed with result 'exit-code'.

vmatare, 2017-02-06:
Hm, it's strange that there's no error being logged. Try running `journalctl -xe` as the message states after trying to start `thinkfan`. If that doesn't reveal any error message as well, try running thinkfan directly on the terminal: `thinkfan -n`. That should definitely give some kind of hint about what's going wrong.

arooni, 2017-02-07:
Thanks for the quick response; 

You were right; there was a problem with my config file.  Which I fixed; but I'm still experiencing the problem in getting it to run automatically.  It appears to run fine if I do sudo thinkfan ;; (Daemon PID: 22285) ;; but then how can I have this run on startup?

`

[I]  ✘  ~  sudo thinkfan -n
[sudo] password for david:

/etc/thinkfan.conf:43:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:44:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:45:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:46:sensor /sys/devices/virtual/hwmon/hwmon0/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:57:(127,     79,     32767)
LOWER limit doesn't overlap with previous UPPER limit.
Refusing to run without usable config file!
[I]  ✘  ~  sudo vim /etc/thinkfan.conf
[I]  ~  sudo thinkfan -n

/etc/thinkfan.conf:43:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:44:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:45:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:46:sensor /sys/devices/virtual/hwmon/hwmon0/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

WARNING: Using default fan control in /proc/acpi/ibm/fan.

WARNING: You're using simple temperature limits without correction values, and your fan will only start at 55 °C. This can be dangerous for your hard drive.

sleeptime=5, tmax=56, last_tmax=56, biased_tmax=56 -> fan="level 3"

^C
Caught deadly signal. Cleaning up and resetting fan control.
[I]  ~  systemctl status thinkfan.service
● thinkfan.service - simple and lightweight fan control program
   Loaded: loaded (/lib/systemd/system/thinkfan.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-02-06 13:50:27 CST; 6h ago
  Process: 1153 ExecStart=/usr/sbin/thinkfan (code=exited, status=4)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
[I]  ✘  ~  sudo thinkfan -n

/etc/thinkfan.conf:43:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:44:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:45:sensor /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

/etc/thinkfan.conf:46:sensor /sys/devices/virtual/hwmon/hwmon0/temp1_input
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

WARNING: Using default fan control in /proc/acpi/ibm/fan.

WARNING: You're using simple temperature limits without correction values, and your fan will only start at 55 °C. This can be dangerous for your hard drive.

sleeptime=5, tmax=55, last_tmax=55, biased_tmax=55 -> fan="level 3"
^C
Caught deadly signal. Cleaning up and resetting fan control.
[I]  ~ 
`

arooni, 2017-02-07:
ps; my config file is here: https://gist.github.com/d9475f5b6926f8b5cd40222037f9e8c6

vmatare, 2017-02-14:
Well, once you fixed your config file, everything should be fine. I don't use Ubuntu, so I don't know how you're supposed to enable a service on bootup there. However on any halfways compliant Linux that runs systemd, it should suffice to enable the systemd service:

`systemctl enable thinkfan`

arooni, 2017-02-14:
It's working now!  Thanks for your help... what a great piece of software you've written.  You can close out this issue.

closed 2017-02-14 by arooni