vmatare/thinkfan #33: ERROR: TpSensorDriver

Sulter, 2017-03-14:
When I'm trying to run thinkfan I get the following message:
```
root@sulter:~# thinkfan -c /etc/thinkfan.conf

ERROR: TpSensorDriver: Initializing sensor in /proc/acpi/ibm/thermal: Succes
```
I'm using the default config file:

> (0,     0,      55)
> (1,     50,     60)
> (2,     55,     61)
> (3,     56,     63)
> (4,     57,     65)
> (5,     59,     66)
> (7,     63,     32767)

I'm able to manipulate /proc/acpi/ibm/thermal manually without any problems.

vmatare, 2017-03-14:
Which version are you using?

Sulter, 2017-03-14:
Commit 3b4589cb31e95c48327750f006847f00e32c2668
The newest version.
I guess I should try the release instead?

Sulter, 2017-03-14:
I can confirm that 1.0 beta3 has the same issue, while 1.0 beta1 works just fine.

knochenhans, 2017-04-08:
Hi, not sure if that helps, but I got exactly the same behaviour here (L420), so latest release gives me the TpSensorDriver error while 1.0 beta1 works without problems.

vmatare, 2017-04-15:
Dumb question: Does `/proc/acpi/ibm/thermal` even exist? I'm asking because on recent systems, it usually doesn't exist any more...

knochenhans, 2017-04-15:
It exists for me (using kernel 4.10.8):
`cat /proc/acpi/ibm/thermal                                   
temperatures:	47 0 47 0 31 54 24 54`

vmatare, 2017-04-16:
Well, sorry it took so long, but to test this I actually had to dig an old T42 from the shelves ;-)
Anyways, the problem should be fixed now in the current master.

vmatare, 2017-09-08:
Closing for lack of feedback after the fix was pushed.

closed 2017-09-08 by vmatare