vmatare/thinkfan #34: Lever oscillation

diggit, 2017-03-29:
Hi,
I have problem with constant level oscillation.

Version:
`thinkfan 0.99.0`

Here is my config:

```
hwmon /sys/class/hwmon/hwmon2/temp1_input           #1
tp_fan /proc/acpi/ibm/fan

{ "level auto"
    (30)
    (85)
}

{ "level disengaged" # nice idea: "level auto" can also be used.
                     # but again: only numbers for sysfs.
    (75)
    (110)
}
```
Output:
```
# thinkfan -s1 -n -d -v
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 96(0) -> level auto
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 98(0) -> level auto
Temperatures(bias): 97(0) -> level disengaged
Temperatures(bias): 97(0) -> level auto
Temperatures(bias): 97(0) -> level disengaged
Temperatures(bias): 96(0) -> level auto
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 97(0) -> level auto
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 97(0) -> level auto
Temperatures(bias): 97(0) -> level disengaged
Temperatures(bias): 96(0) -> level auto
Temperatures(bias): 91(0) -> level disengaged
Temperatures(bias): 89(0) -> level auto
Temperatures(bias): 89(0) -> level disengaged
Temperatures(bias): 95(9) -> level auto
Temperatures(bias): 97(6) -> level disengaged
Temperatures(bias): 95(0) -> level auto
Temperatures(bias): 95(0) -> level disengaged
Temperatures(bias): 95(0) -> level auto
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 96(0) -> level auto
Temperatures(bias): 98(0) -> level disengaged
Temperatures(bias): 98(0) -> level auto
Temperatures(bias): 96(0) -> level disengaged
Temperatures(bias): 98(0) -> level auto
and so on....
```

I tried to tune `-b` parameter, but it had no effect. Is there something wrong with my configuration or what is happening?

vmatare, 2017-04-15:
Hi, thx for the report & sorry for the late reply. I just updated the `master` branch with a bunch of new features & bugfixes. Your particularly issue should be addressed by commit #276554c. Please try with the latest master branch and see if works now.

diggit, 2017-04-15:
Awesome, everything works well. BTW new yaml config example is missing in CMakeLists.txt.

closed 2017-04-15 by diggit

diggit, 2017-04-15:
Using new yaml config, I am not able to set tp_fan:
```
sensors:
  - hwmon: /sys/class/hwmon/hwmon2
    indices: [1]

fans:
  - tp_fan: /proc/acpi/ibm/fan

#
# Complex level config
#
levels:
  - speed: level auto
    upper_limit: [85]

  - speed: level disengaged
    lower_limit: [75]
```

results in:
```
ERROR: /etc/thinkfan.conf.yaml:6:
  - tp_fan: /proc/acpi/ibm/fan
    ^
Invalid entry.
```

What is wrong?

vmatare, 2017-04-15:
Oh right, sorry, the syntax was changed, check out the update `thinkfan.conf.yaml` example ;-)

diggit, 2017-04-15:
Thanks. Everything works for me.
Can you update CMakeLists.txt to install yaml config example too?