vmatare/thinkfan #35: show hard disc temperature sensor by `cat /path/to/harddisk's/temp_input`

Konfekt, 2017-04-19:
The temperature of the hard disc can be shown by
```sh
  sudo smartctl -A /dev/sda | grep Temperature
```
Can it also be read off by a `cat /path/to/harddisk's/temp_input` so that thinkfan can avoid overheating ?

vmatare, 2017-04-20:
It's not clear what you're asking. Avoid overheating of what? And why not just use the built-in libatasmart interface?

Konfekt, 2017-04-20:
In `thinkfan.conf.sysfs` there's a remark
```
# !! ATTENTION !!
#
# To keep your HD from overheating, you have to specify a correction value for
# the sensor that has the HD's temperature. You need to do this because
# thinkfan uses only the highest temperature it can find in the system, and
# that'll most likely never be your HD, as most HDs are already out of spec
# when they reach 55 °C.
#
```
Comparing the output of
```
sudo smartctl -A /dev/sda | grep Temperature

194 Temperature_Celsius     0x0022   063   047   ---    Old_age   Always       -       37 (Min/Max 16/47
```
with that of
```sh
find -L /sys/class/hwmon -maxdepth 5 -name "temp*_input" -print -exec cat \{\} \; 2>/dev/null

/sys/class/hwmon/hwmon0/device/temp3_input
0
/sys/class/hwmon/hwmon0/device/temp4_input
0
/sys/class/hwmon/hwmon0/device/temp5_input
0
/sys/class/hwmon/hwmon0/device/temp6_input
0
/sys/class/hwmon/hwmon0/device/temp7_input
0
/sys/class/hwmon/hwmon0/device/temp1_input
51000
/sys/class/hwmon/hwmon0/device/temp8_input
0
/sys/class/hwmon/hwmon0/device/temp2_input
0
/sys/class/hwmon/hwmon1/temp1_input
51000
/sys/class/hwmon/hwmon2/temp4_input
48000
/sys/class/hwmon/hwmon2/temp2_input
49000
```
it follows that there is no sensor of the format required by thinkfan to read off the HD temperature. 

How to configure thinkfan so that it reads off the HD temperature to specify the correction value? By the above, the way the README proposes does not work. Instead your suggestion is to use the libatasmart interface? How?

vmatare, 2017-09-08:
That is correct, there is no hwmon file in sysfs that represents the SMART temperature sensor. Thinkfan reads it directly from the hard disk, just like `smartctl` does. Please read the THERMAL SENSORS section in `thinkfan.conf(5)`.

Konfekt, 2017-09-08:
Looking at

https://github.com/vmatare/thinkfan/blob/master/examples/thinkfan.conf.simple

and the example
```sh
# I use this on my T61p:
# tp_fan /proc/acpi/ibm/fan
# tp_thermal /proc/acpi/ibm/thermal (0, 10, 15, 2, 10, 5, 0, 3, 0, 3)
```
it would be interesting to have an idea on how high the correction should usually be, for example for an SSD and a laptop CPU.

Konfekt, 2017-09-08:
Looking at

```sh
# IMPORTANT:
#
# To keep your HD from overheating, you have to specify a correction value for
# the sensor that has the HD's temperature. You need to do this because
# thinkfan uses only the highest temperature it can find in the system, and
# that'll most likely never be your HD, as most HDs are already out of spec
# when they reach 55 °C.
# Correction values are applied from left to right in the same order as the
# temperatures are read from the file.
#
# For example:
# tp_thermal /proc/acpi/ibm/thermal (0, 0, 10)
# will add a fixed value of 10 °C the 3rd value read from that file. Check out
# http://www.thinkwiki.org/wiki/Thermal_Sensors to find out how much you may
# want to add to certain temperatures.
#
# I use this on my T61p:
# tp_fan /proc/acpi/ibm/fan
# tp_thermal /proc/acpi/ibm/thermal (0, 10, 15, 2, 10, 5, 0, 3, 0, 3)
```

Does this mean that `/proc/acpi/ibm/thermal` has 7 sensors that measure the HD temperature?

Here, on a X200, there's only one entry, the first one, that measures the CPU temperature.

Konfekt, 2019-02-04:
Do you mean using the new  complex mode?