vmatare/thinkfan #37: thermald

Konfekt, 2017-04-20:
Could you compare `thinkfan` to `thermald`? Does `thermald` provide a superset of `thinkfan` in that it scales CPU frequency and fan speed?

See for example 

https://smackerelofopinion.blogspot.com.br/2015/09/tweaking-thermald-configuration-file.html

vmatare, 2017-04-20:
As far as I know, thermald uses a different fan control mechanism, i.e. there's no direct, user-specified mapping between temperatures and fan speed. There's probably some PID-like algorithm going on in the background, if I'm interpreting the config semantics correctly. Of course thinkfan doesn't manage clock speeds and idle states, so in that regard, thermald implements a superset. It's also perfectly feasible to run thermald alongside thinkfan (as long as fan control is disabled in thermald), since the detailed, direct control offered by thinkfan's Complex Mode might be beneficial in certain scenarios.

Konfekt, 2017-04-21:
> As far as I know, thermald uses a different fan control mechanism, i.e. there's no direct, user-specified mapping between temperatures and fan speed.

The config file https://gist.github.com/ValdikSS/d896cebd8e1aafc65b4b seems to define fan speeds for temperature zones by modifing the pwm values.