vmatare/thinkfan #38: level disengaged when using pwm_fan

Konfekt, 2017-04-20:
The thinkpad_acpi interface with /proc/acpi/ibm/fan has a level `disengaged` with value `127` for maximal fan speed. Does the PWM fan control file offer something similar? It seems that `255` for the PWM fan control file corresponds to level `7` of the thinkpad_acpi interface with /proc/acpi/ibm/fan.

vmatare, 2017-04-20:
No, as far as I know the sysfs PWM interface doesn't offer anything equivalent to `level disengaged`, I'd say that's due to the definition of the interface that only accomodates a linear PWM scale and no special values. But one would have to look at the `thinkpad_acpi` module's source code to know for sure ;-)

Konfekt, 2017-04-21:
Is there any good reason not to use `disengaged` for critical temperatures? That seems to make the `thinkpad_acpi` the only sensible choice, considering that most turn to thinkfan after their thinkpad shut down due to overheating.

vmatare, 2017-04-26:
Yeah, I've seen some Thinkpads overheat as well, mainly the T*s models. I think with these models Lenovo's idea is to reduce clock speed when they overheat, which is what `thermald` does.
That said, if you want to keep your clock speeds maxed out, at the cost of some additional wear on your fan, `thinkfan` and `level disengaged` are the way to go. I don't mind having to replace the fan now and then, so that's what I do ;-)

However, `level disengaged` should be used only as a last resort. On most models, the firmware doesn't even use `level 7`, so even that should help a good bit against overheating.

Konfekt, 2017-04-26:
Interesting, never thought about replacing the fan of my thinkpad. How frequently do you replace your fan now and then? Mine is almost 7 years old. How do you judge a replacement necessary?

diggit, 2017-04-26:
BTW I use _level disengaged_ because I swapped my CPU for quad core (8 thread) in my T430. Heatsink is rated for 35W, but new i7 can produce up to 45W of heat. 

(Many people say, that GPU version heatsink is better. I can't agree, there is additional heatpipe for GPU, but no increased surface to dissipate more heat.)

To maintain temperature below 100°C in dock running compilation in 8 threads, disengaged is best.

ngkz, 2019-05-31:
For most driver, Writing `0` to `pwm*_enable` is equivalent to `level disengaged`.

See https://github.com/torvalds/linux/blob/master/Documentation/hwmon/sysfs-interface.rst#pwm and https://github.com/torvalds/linux/blob/v5.1/drivers/platform/x86/thinkpad_acpi.c#L8496

vmatare, 2020-04-08:
> Interesting, never thought about replacing the fan of my thinkpad. How frequently do you replace your fan now and then? Mine is almost 7 years old. How do you judge a replacement necessary?

I think the fan bearings in my T530p started to fail after ~5 years. You generally notice because it starts getting noisy (rattling, humming, squeaking...). But I'm a Gentoo user and a C++ developer, so there's always a lot of compiling going on with the fan on full trottle.

jdchristensen, 2022-01-23:
@vmatare According to the thinkpad_acpi.c source code, you can get disengaged mode by writing 0 to the pwm1_enable file:
```
 * pwm*_enable:
 * 	0: "disengaged" mode
 * 	1: manual mode
 * 	2: native EC "auto" mode (recommended, hardware default)
```
I just tested this, and it works on my X1 Extreme Gen 4.