vmatare/thinkfan #39: Wifi card turn off -> thinkfan gives error and exits

ilkerhk, 2017-05-26:
I observe the following error in two scenarios : 
1- If I disable wifi manually to save battery
2- Thinkpad bios has option to disable wifi card hardware if ethernet connected

Then the hwmon file becomes unreadable and therefore thinkfan gives the following error and exits: 

thinkfan[19319]: /sys/devices/virtual/hwmon/hwmon4/temp1_input: Input/output error
thinkfan[19319]: A sensor has vanished! Exiting since there's no safe way of handling this.

I wanted to exclude wifi temperature sensor from the .conf file but the virtual hwmonXX changes in every boot. 

One nice solution to this (and also to changing hwmonX folders)  would be to replace the sensors with device-names in conf file. The conf file would use something like : 
device pch_skylake-virtual-0
instead of 
hwmon /sys/devices/virtual/hwmon/hwmon4/temp1_input

I have no idea if that is possible or easy to develop...

ilkerhk, 2017-05-27:
I worked on an ugly workaround.  Now autostart of thinkfan service is disabled and it is started after user logs in, by a script called thinkfanLauncher.sh. It creates the conf file first  (coppied below). In my machine this workaround solves following issues:

1- locate the hwmon folders (they change after each boot)  
2- excluding a device by its name. (I exclude wifi card so that auto or manual disabling of it doesn't stop thinkfan)

code coppied below. Hopefully there will be a permanent solution of these in thinkfan...

 `
######## stop previous #####################################################
systemctl stop thinkfan 
sleep 1 

######## Create conf #######################################################
fileName="/etc/thinkfan.conf"

echo " 
# This file is automatically generated by thinkfanLauncher.sh. 
# 1- The sensors : 
"   > $fileName
 
#space separater list  
excludeList="iwlwifi" 

for sensorFolder in $(find /sys/class/hwmon) 
do 
  name=$(cat $sensorFolder/name  2> /dev/null) 
  for sensor in $(find $sensorFolder/ -type f -name "temp*_input") 
  do 
    if [[ $excludeList == *$name* ]]
    then
      echo "Excluding $name (in the exclude list)";
    else
      cat $sensor &> /dev/null > /dev/null
      if [ $? -eq 0 ]
      then 
        echo !!! $name : Found a readable temp sensors. Adding to config. 
        echo "hwmon $sensor" >> $fileName
      else
        echo "Excluding $name ( has a temp sensor but unreadable. Maybe turned off??) " 
      fi
    fi
  done
done

echo "# 2- the fan : 
tp_fan /proc/acpi/ibm/fan   
# 3- The temperature scheme : 
(0,	0,	58)    
(1,	45,	66)   
(2,	65,	72)   
(3,	71,	74)   
(4,	73,	76)   
(5,	75,	78)   
(7,	77,	80)   
"   >> $fileName

######## Restart thinkfan ##################################################
# /usr/sbin/thinkfan -b -10 -c /etc/thinkfan.conf
systemctl start thinkfan 
`

vmatare, 2017-07-03:
Well, I'm not sure whether I just want to ignore whether a sensor has disappeared because I can imagine situations where that might lead to damage. On the other hand, if you enable it via a commandline option, it's fine...

ploth, 2019-03-03:
I encounter the same problem and need the option for optional sensors. That would be so great! Thank you!

vmatare, 2020-04-08:
anyone who wants this, please try out https://github.com/vmatare/thinkfan/tree/optional-sensors . I need some feedback on how that is working for you so I can make it part of the next release.

vmatare, 2020-04-08:
It's merged in the `master` branch now to give it some more exposition. Hoping for feedback ;-)

mustaphazorgati, 2020-04-26:
Hey there! 
Unfortunately thinkfan breaks on my system (Thinkpad T470p) when I turn of the wifi card.
```
$ sudo thinkfan -n -c /etc/thinkfan.conf.yaml 
Temperatures(bias): 41(0), 41(0), 40(0), 41(0), 40(0), 40(0), 29(0), 29(0), 33(0), 35(0), 32(0) -> level 0
ERROR: read_temps: Failed to read temperature(s) from /sys/devices/virtual/thermal/thermal_zone3/hwmon7/temp1_input: No data available
``` 

thinkfan.conf.yaml
```
sensors:
  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon4
    indices: [1]
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon5
    indices: [1, 2, 3, 4, 5]
  - hwmon: /sys/devices/pci0000:00/0000:00:1d.0/0000:05:00.0/nvme/nvme0/hwmon1
    indices: [1, 2, 3]
  - hwmon: /sys/devices/virtual/thermal
    name: pch_skylake
    indices: [1]
  - hwmon: /sys/devices/virtual/thermal
    name: iwlwifi_1
    indices: [1]
    optional: true
fans:
  - tpacpi: /proc/acpi/ibm/fan
levels:
  - [  0,  0,    42]
  - [  1, 40,    47]
  - [  2, 45,    52]
  - [  3, 50,    57]
  - [  4, 55,    62]
  - [  5, 60,    77]
  - [  7, 73,    93]
  - [127, 85, 32767] 
```

vmatare, 2020-06-05:
Has been implemented in master for a while now, and will be part of upcoming release.

closed 2020-06-05 by vmatare