vmatare/thinkfan #40: yaml interface indexes

ilkerhk, 2017-05-26:
I am trying to use yaml configuration file.  Consider following simplified scenario : 

/sys/class/hwmon/hwmon0/temp1_input
/sys/class/hwmon/hwmon0/temp2_input
/sys/class/hwmon/hwmon1/temp1_input

using index [1,2] 
thinkfan only finds the first two sensors, can't find the last one. I am guessing this might be due to the fact they have the same name?

vmatare, 2017-07-03:
Hi, sorry about the delay. Could you please post your entire config? It might be an indentation problem, yaml is quite picky about that ;-)

ilkerhk, 2017-07-03:
here it is...

–ilker.

vmatare, 2017-07-03:
Huh, sorry, seems like github has removed the email attachment? It should be fine if you copy&paste it using the "insert code" button.

ilkerhk, 2017-07-03:
I copied below. best.

sensors:
  - hwmon: /sys/class/hwmon/hwmon0
    indices: [1,2, 3]
  - hwmon: /sys/class/hwmon/hwmon1
    indices: [1,2, 3]

  #- tpacpi: /proc/acpi/ibm/thermal

fans:
  # - hwmon: /sys/class/graphics/fb0/device/hwmon
  #   indices: [1]
  - tpacpi: /proc/acpi/ibm/fan

#
# Complex level config
#
# levels:
#   - speed: 0
#    upper_limit: [50, 50, 50]
#
#  - speed: 2
#    lower_limit: [45, 45, 45]
#    #upper_limit: [60, 60, 60]

#
# Simple level config
# meaning: [FANSPEED, LOWER_LIMIT, UPPER_LIMIT]
#
#levels:
#  - [0, 0, 50]
#  - [2, 45, 60]

levels:
 - [0, 0, 58]
 - [1, 45, 66]
 - [2, 65, 72]
 - [3, 71, 74]
 - [4, 73, 76]
 - [5, 75, 78]
 - [7, 77, 80]

–ilker.

diggit, 2017-07-04:
If I understand yaml config properly. There should be something like this for your "simplified scenario" described in 1st post.

```
sensors:
  - hwmon: /sys/class/hwmon/hwmon0
    indices: [1,2]
  - hwmon: /sys/class/hwmon/hwmon1
    indices: [1]
...
```
Just a quick idea. Not tested or so.

ilkerhk, 2017-07-04:
Yes, I fothot to say that the config I sent is not for simplified scenario.
This is the one for my machine which has all 3 sensors.

vmatare, 2018-07-16:
I've not been able to reproduce this. If this is still an issue, please post which version you're using (`git show HEAD`) and the output of `thinkfan -nvv`.

Closing for now since it's the only issue holding back the 1.0 release.

closed 2018-07-16 by vmatare

menobista, 2020-04-15:
Hi, I am new to thinkfan. I was also having the same with hwmon sensors path issue. I made this config file.
When i ran the below config it shows.
./thinkfan.yaml:0:sensors:
WARNING: The `sensor' keyword is deprecated. Please use the `hwmon' or `tp_thermal' keywords instead!

./thinkfan.yaml:2:- hwmon: /sys/devices/platform/coretemp.0/hwmon
   
Syntax error.
Refusing to run without usable config file!

Any help
HERE IS MY CONFIG FILE
sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [3, 4, 1, 5, 2]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - speed: 0
    lower_limit: [0, 0, 0]
    upper_limit: [59, 59, 59]

  - speed: 1
    lower_limit: [52, 52, 52]
    upper_limit: [75, 75, 75]

  - speed: 2
    lower_limit: [65, 65, 65]
    upper_limit: [78, 78, 78]

  - speed: 3
    lower_limit: [66, 66, 66]
    upper_limit: [76, 76, 76]

  - speed: 4
    lower_limit: [67, 67, 67]
    upper_limit: [82, 82, 82]

  - speed: 5
    lower_limit: [68, 68, 68]
    upper_limit: [85, 85, 85]

  - speed: 7
    lower_limit: [75, 75, 75]
    upper_limit: [32767, 32767, 32767]

vmatare, 2020-04-16:
@menobista please don't hijack random existing issues. Your comment has nothing to do with this issue. If you're not exactly sure what your problem is, it's better to open a new issue.