vmatare/thinkfan #41: Crash when sensor is unreadable

rdoursenaud, 2017-08-28:
I have several sensor paths declared.
Unfortunately, one (iwlwifi) is not readable since some kernel update for an undetermined reason.

When starting Thinkfan crashes with the trace below.

I'd expect it to issue warning about the unavailable sensor and continue.

```
ERROR: Unhandled std::ios_base::failure[abi:cxx11]: basic_filebuf::underflow error reading the file: iostream error.
errno = Input/output error.
Backtrace:
/usr/sbin/thinkfan(+0x37764) [0x55c6b0118764]
                                             /usr/sbin/thinkfan(thinkfan::handle_uncaught()+0x20b) [0x55c6b0119acb]
                                             /usr/lib/libstdc++.so.6(+0x8fc46) [0x7f8bfac95c46]
                                             /usr/lib/libstdc++.so.6(+0x8fc91) [0x7f8bfac95c91]
                                             /usr/lib/libstdc++.so.6(+0x8fed3) [0x7f8bfac95ed3]
                                             /usr/lib/libstdc++.so.6(std::__throw_ios_failure(char const*)+0x63) [0x7f8bfacc0a83]
                                             /usr/lib/libstdc++.so.6(std::basic_filebuf<char, std::char_traits<char> >::underflow()+0x39f) [0x7f8bfacf58bf]
                                             /usr/lib/libstdc++.so.6(std::istream::sentry::sentry(std::istream&, bool)+0x19a) [0x7f8bfad0206a]
                                             /usr/lib/libstdc++.so.6(std::istream::operator>>(int&)+0x2e) [0x7f8bfad0224e]
                                             /usr/sbin/thinkfan(thinkfan::HwmonSensorDriver::read_temps() const+0x176) [0x55c6b0108746]
                                             /usr/sbin/thinkfan(thinkfan::run(thinkfan::Config const&)+0x7d) [0x55c6b00fc6dd]
                                             /usr/sbin/thinkfan(main+0x422) [0x55c6b00fadd2]
                                             /usr/lib/libc.so.6(__libc_start_main+0xea) [0x7f8bfa6694ca]
                                             /usr/sbin/thinkfan(_start+0x2a) [0x55c6b00fb9ca]
```

```/etc/thinkfan.conf```
```
sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon0/temp1_input  # CPU Package
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon0/temp2_input  # CPU Core 0
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon0/temp3_input  # CPU Core 1
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon0/temp4_input  # CPU Core 2
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon0/temp5_input  # CPU Core 3
  - hwmon: /sys/devices/virtual/hwmon/hwmon2/temp1_input  # pch_skylake
  - hwmon: /sys/devices/virtual/hwmon/hwmon3/temp1_input  # acpitz (ACPI CPU)
#  - hwmon: /sys/devices/virtual/hwmon/hwmon4/temp1_input  # iwlwifi
  - nvidia: 01:00.0  # nVidia GPU
  - atasmart: /dev/sda  # HDD
    correction: [10]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0,  40]
  - [1, 32, 55]
  - [2, 54, 66]
  - [3, 65, 76]
  - [4, 75, 80]
  - [5, 78, 85]
  - [6, 82, 88]
  - [7, 85, 32767]
```

rdevulap, 2017-10-25:
May i know which thinkpad you are using @rdoursenaud ?

rdoursenaud, 2017-10-26:
Sure, I'm using a P51.
I've published the full specs and some notes on my blog: https://raphael.doursenaud.fr/lenovo-thinkpad-p51.html

In the meantime, I've noticed that this behavior is consistent with wifi being (software) disabled.
The iwlwifi module is still loaded and the hwmon path present but it returns an I/O error.
Other software trying to read it (like sensors or glances) issue a warning but don't crash.

rdevulap, 2017-10-26:
Awesome, thanks! I was thinking of buying a thinkpad, but wasn't sure if Lenevo still lets users control the fan speed. Seems like they do! :)

rdoursenaud, 2017-11-22:
Yes, it does :)

rdevulap, 2017-11-22:
Awesome, i just bought one too. I installed OpenSuse though and thinkpad_acpi driver seems to have some issues. I will switch to Arch linux and hopefully it will be okay.

vmatare, 2018-05-21:
Hi, thanks for the report and for posting the backtrace, specifically. Having optional sensors is a feature currently in development, and this uncaught exception would have been a showstopper for certain ;-)

vmatare, 2018-05-21:
I just pushed the `optional-sensors` branch which should address this & related issues. Would be highly appreciated if you (or anyone) could test how it behaves in these situations.

vmatare, 2020-04-08:
anyone who wants this, please try out the [optional-sensors]( https://github.com/vmatare/thinkfan/tree/optional-sensors) branch. I need some feedback on how that is working for you so I can make it part of the next release.

vmatare, 2020-04-08:
It's merged in the `master` branch now to give it some more exposition. Hoping for feedback ;-)

vmatare, 2020-06-05:
Has been implemented in master for a while now, and will be part of upcoming release.

closed 2020-06-05 by vmatare