vmatare/thinkfan #42: Compile error in thinkfan::Config::try_read_config()

mathsian, 2017-10-13:
Hi. 

thinkfan 1.0 
libyaml-cpp0.5v5
libyaml-cpp-dev

```
src/config.cpp:75:49: error: passing ‘const YAML::wtf_ptr<thinkfan::Config>’ as ‘this’ argument discards qualifiers [-fpermissive]
   rv = root.as<YAML::wtf_ptr<Config>>().release();
                                                 ^
In file included from src/config.cpp:30:0:
src/yamlconfig.h:40:7: note:   in call to ‘auto YAML::wtf_ptr<T>::release() [with T = thinkfan::Config]’
  auto release()
       ^
CMakeFiles/thinkfan.dir/build.make:86: recipe for target 'CMakeFiles/thinkfan.dir/src/config.cpp.o' failed
```

mathsian, 2017-10-13:
Probably more helpful is
`gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5)`

vmatare, 2017-10-13:
Huh. That sucks. Seems like another case of Ubuntu shipping ancient software. This is a bug in libyaml-cpp that has been fixed in 2015. Cf. jbeder/yaml-cpp@97d56c3f3608331baaee26e17d2f116d799a7edc

vmatare, 2017-10-13:
I think I can work around it, but could you do me and the rest of the world a favor and report this as a Ubuntu bug so that maybe they consider updating libyaml-cpp to something that isn't 2 years old?

mathsian, 2017-10-13:
More than happy to do that. And no rush for a work around. I can find or build a newer libyaml-cpp, I'm sure.

vmatare, 2017-10-13:
nah, don't do that, the workaround is rather trivial. I'd just like to see Ubuntu bothered since they keep annoying developers with their outdated ecosystem. However 16.04 is widespread enough so I do want to support it out of the box, regardless of whether they fix their libyaml-cpp.
Anyways, thanks for the report. Please retry with the updated master (cf. 7efacbf)

mathsian, 2017-10-13:
Sorry, `config.cpp` compiled, but now

```
src/error.cpp:158:11: error: ‘const struct YAML::Mark’ has no member named ‘is_null’
  if (mark.is_null())
```

mathsian, 2017-10-13:
Oh, I think it should just be `null_mark`

vmatare, 2017-10-13:
No, there's actually a method missing there. Workaround is in the current master. Thanks for bearing with me, I hope there isn't too many of those left ;-)

mathsian, 2017-10-13:
I worked around `is_null` by checking for `mark.pos == -1` but still get

```
src/yamlconfig.cpp:184:47: error: no matching function for call to ‘find_hwmons(thinkfan::string&, const std::vector<int>)’
       path, node[kw_indices].as<vector<int>>());
```

mathsian, 2017-10-13:
That output in full 
```[ 12%] Building CXX object CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o
/home/ian/thinkfan/src/yamlconfig.cpp: In static member function ‘static bool YAML::convert<std::vector<YAML::wtf_ptr<thinkfan::HwmonSensorDriver> > >::decode(const YAML::Node&, std::vector<YAML::wtf_ptr<thinkfan::HwmonSensorDriver> >&)’:
/home/ian/thinkfan/src/yamlconfig.cpp:184:47: error: no matching function for call to ‘find_hwmons(thinkfan::string&, const std::vector<int>)’
       path, node[kw_indices].as<vector<int>>());
                                               ^
/home/ian/thinkfan/src/yamlconfig.cpp:110:27: note: candidate: template<class T> std::vector<YAML::wtf_ptr<T> > YAML::find_hwmons(thinkfan::string, std::vector<int>&&)
 static vector<wtf_ptr<T>> find_hwmons(string path, vector<int> &&indices)
                           ^
/home/ian/thinkfan/src/yamlconfig.cpp:110:27: note:   template argument deduction/substitution failed:
/home/ian/thinkfan/src/yamlconfig.cpp:184:45: note:   cannot convert ‘YAML::Node::as() const [with T = std::vector<int>]()’ (type ‘const std::vector<int>’) to type ‘std::vector<int>&&’
       path, node[kw_indices].as<vector<int>>());
                                             ^
/home/ian/thinkfan/src/yamlconfig.cpp:187:39: error: ‘const class YAML::Node’ has no member named ‘Mark’
      throw YamlError(node[kw_indices].Mark(), MSG_CONF_CORRECTION_LEN(
                                       ^
/home/ian/thinkfan/src/yamlconfig.cpp: In static member function ‘static bool YAML::convert<std::vector<YAML::wtf_ptr<thinkfan::SensorDriver> > >::decode(const YAML::Node&, std::vector<YAML::wtf_ptr<thinkfan::SensorDriver> >&)’:
/home/ian/thinkfan/src/yamlconfig.cpp:277:25: error: ‘const class YAML::Node’ has no member named ‘Mark’
    throw YamlError(node.Mark(), "Sensor entries must be a sequence. Forgot the dashes?");
                         ^
/home/ian/thinkfan/src/yamlconfig.cpp:280:85: error: binding ‘const YAML::wtf_ptr<thinkfan::HwmonSensorDriver>’ to reference of type ‘YAML::wtf_ptr<thinkfan::HwmonSensorDriver>&’ discards qualifiers
     for (wtf_ptr<HwmonSensorDriver> &h : it->as<vector<wtf_ptr<HwmonSensorDriver>>>())
                                                                                     ^
/home/ian/thinkfan/src/yamlconfig.cpp:283:56: error: no matching function for call to ‘std::vector<YAML::wtf_ptr<thinkfan::SensorDriver> >::push_back(const YAML::wtf_ptr<thinkfan::TpSensorDriver>)’
     sensors.push_back(it->as<wtf_ptr<TpSensorDriver>>());
                                                        ^
In file included from /usr/include/c++/5/vector:64:0,
                 from /home/ian/thinkfan/src/thinkfan.h:25,
                 from /home/ian/thinkfan/src/message.h:30,
                 from /home/ian/thinkfan/src/error.h:28,
                 from /home/ian/thinkfan/src/config.h:25,
                 from /home/ian/thinkfan/src/yamlconfig.h:4,
                 from /home/ian/thinkfan/src/yamlconfig.cpp:1:
/usr/include/c++/5/bits/stl_vector.h:913:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = YAML::wtf_ptr<thinkfan::SensorDriver>; _Alloc = std::allocator<YAML::wtf_ptr<thinkfan::SensorDriver> >; std::vector<_Tp, _Alloc>::value_type = YAML::wtf_ptr<thinkfan::SensorDriver>]
       push_back(const value_type& __x)
       ^
/usr/include/c++/5/bits/stl_vector.h:913:7: note:   no known conversion for argument 1 from ‘const YAML::wtf_ptr<thinkfan::TpSensorDriver>’ to ‘const value_type& {aka const YAML::wtf_ptr<thinkfan::SensorDriver>&}’
/usr/include/c++/5/bits/stl_vector.h:931:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = YAML::wtf_ptr<thinkfan::SensorDriver>; _Alloc = std::allocator<YAML::wtf_ptr<thinkfan::SensorDriver> >; std::vector<_Tp, _Alloc>::value_type = YAML::wtf_ptr<thinkfan::SensorDriver>]
       push_back(value_type&& __x)
       ^
/usr/include/c++/5/bits/stl_vector.h:931:7: note:   no known conversion for argument 1 from ‘const YAML::wtf_ptr<thinkfan::TpSensorDriver>’ to ‘std::vector<YAML::wtf_ptr<thinkfan::SensorDriver> >::value_type&& {aka YAML::wtf_ptr<thinkfan::SensorDriver>&&}’
/home/ian/thinkfan/src/yamlconfig.cpp:286:58: error: no matching function for call to ‘std::vector<YAML::wtf_ptr<thinkfan::SensorDriver> >::push_back(const YAML::wtf_ptr<thinkfan::NvmlSensorDriver>)’
     sensors.push_back(it->as<wtf_ptr<NvmlSensorDriver>>());
                                                          ^
In file included from /usr/include/c++/5/vector:64:0,
                 from /home/ian/thinkfan/src/thinkfan.h:25,
                 from /home/ian/thinkfan/src/message.h:30,
                 from /home/ian/thinkfan/src/error.h:28,
                 from /home/ian/thinkfan/src/config.h:25,
                 from /home/ian/thinkfan/src/yamlconfig.h:4,
                 from /home/ian/thinkfan/src/yamlconfig.cpp:1:
/usr/include/c++/5/bits/stl_vector.h:913:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = YAML::wtf_ptr<thinkfan::SensorDriver>; _Alloc = std::allocator<YAML::wtf_ptr<thinkfan::SensorDriver> >; std::vector<_Tp, _Alloc>::value_type = YAML::wtf_ptr<thinkfan::SensorDriver>]
       push_back(const value_type& __x)
       ^
[... cut ...]
/home/ian/thinkfan/src/yamlconfig.cpp:479:30: error: ‘const class YAML::Node’ has no member named ‘Mark’
    throw YamlError(it->first.Mark(), "Unknown keyword");
                              ^
CMakeFiles/thinkfan.dir/build.make:206: recipe for target 'CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o' failed
make[2]: *** [CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/thinkfan.dir/all' failed
make[1]: *** [CMakeFiles/thinkfan.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

vmatare, 2017-10-13:
Ah thanks, that one-liner looked a little strange to me. May be a lengthy error, but the answer is in there somewhere ;-)

mathsian, 2017-10-13:
Well out of my depth now, I'm afraid :)

vmatare, 2017-10-13:
The reason it looks so frightening is that it's actually a whole bunch of errors, and I reckon it might be all of them.

mathsian, 2017-10-13:
There's an existing bug report asking for 0.5.3 in Debian [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=847099).

I found packages for `yaml-cpp` 0.5.3 in a [ppa](https://launchpad.net/~winterheart/+archive/ubuntu/openxcom), and latest master built fine against them. Seems to be working perfectly.

vmatare, 2017-10-13:
Thanks a lot for working through this mess. Seeing as both Ubuntu and Debian seem to be moving at the speed of a dead snail moulded in concrete, I'll still try to come up with workarounds. At least no I know I can test against 0.5.2 ;-)

mathsian, 2017-10-13:
No problem. Thank *you* for keeping my laptop alive :)

vmatare, 2017-10-17:
I have updated the master with workarounds for all functionality that is missing in yaml-cpp-0.5.2. Should now compile with Ubuntu 16.04 out of the box. It should output a warning that error messages about logical config errors (not syntax) will be less informative.

mathsian, 2017-10-28:
Closing this because I can't test it any more

closed 2017-10-28 by mathsian