vmatare/thinkfan #43: Support for thermal_zone temperatures by name

jimis, 2017-11-28:
Feature request:

I'm using `thermal_zone` temperatures instead of `hwmon` because the paths are more predictable and do not disappear (unlike the `temp[0-9]_input` hwmon files). Here are the paths I'm using:
```
$ cat /sys/devices/virtual/thermal/thermal_zone[0-3]/temp
43000
40500
30000
44000
```

However they might still change order, so I would like to set each sensor by the name instead of the path. Names are stored in the `type` files:
```
$ cat /sys/devices/virtual/thermal/thermal_zone[0-3]/type
acpitz
pch_skylake
iwlwifi
x86_pkg_temp
```

This way I would adjust the parameters more accurately, for example if I want to exclude a specific sensor (for example the `iwlwifi` that disappears if wifi is disabled).

phrxmd, 2018-12-31:
This would be pretty neat, also because the `thermal_zone` tree usually provides much more fine-grained control over temperature data.

From an UI side It could be made accessible through an extra keyword for temperature inputs in `thinkfan.conf`, in addition to the existing `tp_thermal`, `hwmon` and `atasmart` keywords. E.g. you'd have an extra line:

```
thermal_zone pch_skylake
```

that would give you the temperature input from whatever `thermal_zone*` subtree has  `pch_skylake` in its `type` file.

akheron, 2019-09-24:
I created a PR to fix this: #69

PR #69 (2019-09-25): Add support for searching hwmon sensors by name
Add a new config directive `thermal_zone name` which searches /sys/devices/virtual/thermal for a file called `name`. It its content matches the name, use the `temp1_input` next to it as the temperature input.

Fixes #43
closed 2019-09-25 by vmatare