vmatare/thinkfan #44: Declare a sensor as optional

jimis, 2017-11-28:
Here is my problem, when I disable wifi on my laptop:
```
A sensor has vanished! Exiting since there's no safe way of handling this.
```

Additionally, thinkfan does not even start if this sensor is missing.

So I would like to declare a `hwmon` sensor line as "optional" in the config file.

This combines very nice with issue #43, because I would be able to set the specific sensor for `iwlwifi` as optional.

mkopec, 2018-02-24:
I would also like this to be implemented. Currently, on my laptop (T430 with hybrid graphics), if I add an `nv_thermal` sensor to `thinkfan.conf`, it will fail if the nVidia GPU is disabled (I frequently switch it on and off in the BIOS). If the sensor could be marked as optional, thinkfan could just ignore its absence and continue working normally.

acarapetis, 2018-04-19:
I have the same issue on an E480: whenever the AMD GPU is powered off, the corresponding hwmon sensor returns read errors.

vmatare, 2018-05-21:
Everyone who is interested in this, please take a look at the `optional-sensors` branch and test it. Any feedback is greatly appreciated since I don't have the hardware to actually test this.

acarapetis, 2018-05-22:
Thanks for the effort to implement this. Running your code unmodified, it's currently dying at startup with a read error on the optional sensor: looks like you forgot to guard the IOerror on the initial read (line 115 in thinkfan.cpp). Hacking around this myself, from some brief testing it seems like everything works as intended.

jimis, 2018-06-03:
Are all fixes in the `optional-sensors` branch? I'll try to run a test by next weekend.

jimis, 2018-07-19:
@vmatare this does not compile successfully on RHEL7 (gcc-4.8.5) but master branch compiles fine.

```
[ 16%] Building CXX object CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o

src/thinkfan.cpp: In function ‘void thinkfan::run(const thinkfan::Config&)’:
src/thinkfan.cpp:137:45: error: ‘class std::ios_base::failure’ has no member named ‘code’
     sensor_lost(sensor, IOerror(e.what(), e.code().value()));
                                             ^
make[2]: *** [CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o] Error 1
```

Also I see that the last commit on the branch is on May 21, so I'm guessing that the uncaught exception mentioned in the previous comment is still there. Please let me know when you update the branch and I'll retest.

Nitrooo, 2019-12-15:
What's the current status of this feature? I'm on the master branch and I'd need it for my system (Optimus laptop).

vmatare, 2020-04-08:
anyone who wants this, please try out the [optional-sensors]( https://github.com/vmatare/thinkfan/tree/optional-sensors) branch. I need some feedback on how that is working for you so I can make it part of the next release.

vmatare, 2020-04-08:
It's merged in the `master` branch now to give it some more exposition. Hoping for feedback ;-)

vmatare, 2020-06-05:
Fixed in master for a while now, and will be part of upcoming release.

closed 2020-06-05 by vmatare