vmatare/thinkfan #45: ERROR: Module thinkpad_acpi doesn't seem to support fan_control

pat-s, 2017-12-05:
`sudo systemctl status thinkfan`

```
Dec 05 18:09:05 arch systemd[1]: Starting Thinkfan, the minimalist fan control program...
Dec 05 18:09:05 arch thinkfan[23251]: ERROR: Module thinkpad_acpi doesn't seem to support fan_control
Dec 05 18:09:05 arch systemd[1]: thinkfan.service: Control process exited, code=exited status=1
Dec 05 18:09:05 arch systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Dec 05 18:09:05 arch systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
```

I've set `options thinkpad_acpi fan_control=1` and specified the following config file:

```
### T430
hwmon /sys/devices/virtual/hwmon/hwmon0/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input

(0, 0,  40)
(1, 32, 55)
(2, 54, 66)
(3, 65, 76)
(4, 75, 80)
(5, 78, 85)
(6, 82, 88)
(7, 85, 32767)
```

PS: Applies to both versions 0.9.3 and current git. I'm on a T430.

vmatare, 2018-03-11:
Well, if you don't know what's going on then check it:
```
lsmod | grep thinkpad
cat /proc/acpi/ibm/fan
cat /sys/modules/thinkpad_acpi/parameters/fan_control
```

ferrao, 2019-02-21:
The `experimental=1` option fixed it for me.

s0m3OnE47, 2020-02-04:
> The `experimental=1` option fixed it for me.

How did you do that? Could you elaborate? I am facing same issue

vmatare, 2020-02-04:
It's an option to the `thinkpad_acpi` kernel module. Module options are set by modifying or adding the appropriate files in `/etc/modprobe.d/`. Read the manpage: `man modprobe.d`.

vmatare, 2020-04-10:
I've now put a link to the documentation for the `thinkpad_acpi` kernel module in the "SEE ALSO" section of both man pages. I think that should be sufficient, because if someone isn't paying attention to the documentation, they generally shouldn't be using thinkfan anyways.

closed 2020-04-10 by vmatare

arm3nyn, 2020-05-01:
I got this error when I do : sudo thinkfan -q
ERROR: ~TpFanDriver: Resetting fan control in /proc/acpi/ibm/fan: Invalid argument
ERROR: Module thinkpad_acpi doesn't seem to support fan_control
that mesans not supported for thinkpad e470?

mbernasocchi, 2020-07-15:
> How did you do that? Could you elaborate? I am facing same issue

@s0m3OnE47 you can simply do 
```
echo "options thinkpad_acpi fan_control=1 experimental=1" | sudo tee /etc/modprobe.d/thinkfan.conf
sudo modprobe -rv thinkpad_acpi
sudo modprobe -v thinkpad_acpi
```
this will write the `thinkpad_acpi` options in the `/etc/modprobe.d/thinkfan.conf` and reload the `thinkpad_acpi` module

aabacchus, 2021-05-07:
Just in case anyone is trying this with the `thinkpad_acpi` module built-in, you add the following option to your kernel command 
line:
```
thinkpad_acpi.fan_control=1
```
If you're using grub, add this to `GRUB_CMDLINE_LINUX_DEFAULT` in `/etc/default/grub`, then reconfigure grub with `grub-mkconfig -o /boot/grub/grub.cfg` (or whatever).

queeup, 2022-06-15:
For Fedora Silverblue or Kinoite:
```bash
sudo rpm-ostree kargs --append=thinkpad_acpi.fan_control=1
```