vmatare/thinkfan #46: Thinkfan doesn't start.. noob issue?

RussH, 2018-01-04:
Hi, I replaced the 'legacy' thinkfan with your own to be able to poll the temperature of the NVidia card in my T61p

However it's not starting - is it 'something obvious' from the info below or should I recompile with debug for a more useful output?

When I launch it I get;
`# systemctl start thinkfan
Job for thinkfan.service failed because the control process exited with error code. See "systemctl status thinkfan.service" and "journalctl -xe" for details.`

Info available is;

> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: Starting Thinkfan, the minimalist fan control program...
> Jan 04 14:33:21 russh-ThinkPad-T61 thinkfan[5201]: ERROR: TpSensorDriver: Initializing sensor in /proc/acpi/ibm/thermal: Success
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: thinkfan.service: Control process exited, code=exited status=1
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: thinkfan.service: Unit entered failed state.
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: thinkfan.service: Failed with result 'exit-code'.

and

> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: Starting Thinkfan, the minimalist fan control program...
> -- Subject: Unit thinkfan.service has begun start-up
> -- Defined-By: systemd
> -- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
> -- 
> -- Unit thinkfan.service has begun starting up.
> Jan 04 14:33:21 russh-ThinkPad-T61 thinkfan[5201]: ERROR: TpSensorDriver: Initializing sensor in /proc/acpi/ibm/thermal: Success
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: thinkfan.service: Control process exited, code=exited status=1
> Jan 04 14:33:21 russh-ThinkPad-T61 systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
> -- Subject: Unit thinkfan.service has failed

The config file is the simple conf example, no changes.

vmatare, 2018-01-05:
Hi,
the error handling when dealing with the legacy `/proc/acpi/ibm/thermal` file isn't very nice since I don't have access to older Thinkpad models any more. Please post the contents of that file so I can take a closer look, i.e. do a `cat /proc/acpi/ibm/thermal` and paste the output here.

RussH, 2018-01-05:
Hi @vmatare thanks for looking at this - output of cat /proc/acpi/ibm/thermal is;

```
root@russh-ThinkPad-T61:~# cat /proc/acpi/ibm/thermal
temperatures:	50 32 24 50 25 -128 25 -128 30 35 33 -128 -128 -128 -128 -128
```

vmatare, 2018-01-11:
I can't reproduce the problem here. If you can, please give me a hand with debugging this:
1. Which Linux distribution are you using?
2. Which exact thinkfan version are you using? I.e. paste the output of `git branch -av` here.
3. Compile with developer debugging, i.e. in the main thinkfan directory:
```
mkdir build-debug && cd build-debug
cmake -DCMAKE_BUILD_TYPE=Debug -DDISABLE_EXCEPTION_CATCHING=TRUE ..
make VERBOSE=1
```
Please also post the output of those commands here. The `DISABLE_EXCEPTION_CATCHING` option makes thinkfan consider every error a bug, so it should print a detailed backtrace when the error happens. That's the most important bit. Maybe all of this information will help me figure out the root of your problem ;-)

RussH, 2018-01-12:
Okay - so I'd originally downloaded zip from master and built it - rather than a git checkout. Decided to rebuild with debug after a git clone from master - and of course it built just fine.. So please close this issue as user error!

I've enabled monitoring of the nvidia GPU which seems to work just fine so will comment on your aged issue on monitoring the GPU temp. 

Thanks agaiin!

closed 2018-01-12 by RussH