vmatare/thinkfan #52: /proc/acpi/ibm/thermal doesn't exist T520

The-Soulless, 2018-05-26:
Hello, I'm using a T520 and trying to get Thinkfan to work on Void GNU/Linux. I have copied the simple config but wen I run it I get the following error:
https://hastebin.com/kegavadeku.pas

emmanuelgeoffray, 2018-05-29:
I followed the insturctions there to fix it: http://thinkwiki.de/Thinkfan
```
Nun können die verfügbaren Temperatursensoren mit folgendem Kommando angezeigt werden:

find /sys/devices -type f -name "temp*_input"
Die Ausgabezeilen nimmt mit man einem vorangestellten sensor in /etc/thinkfan.conf auf (vor den Temperaturschwellen).

Beispiel (T420s):

sensor /sys/devices/platform/coretemp.0/temp1_input
sensor /sys/devices/platform/coretemp.2/temp1_input
sensor /sys/devices/virtual/hwmon/hwmon0/temp1_input```

vmatare, 2018-07-15:
Pastebin is empty. Is this still an issue?

DEvil0000, 2018-07-26:
works on my L480 with:
```
hwmon /sys/devices/platform/coretemp.0/temp1_input
...
```

closed 2018-08-24 by vmatare

jorgelaranjo, 2020-05-20:
On a Thinkstation P700 folder `/proc/acpi/ibm/thermal` does not exist. `/sys/devices/platform/coretemp.*/` exist but those don't show any fans nor fan control whatsoever.
Lenovo does say this machine is certified for Ubuntu and CentOS but neither seems to show fan control as my experience shows and the following link [https://askubuntu.com/questions/1238028/how-to-control-fan-speed-on-thinkstation-by-lenovo](shows)

Would be nice to know if fan control is possible under Linux for this platform and which kernel modules need to be loaded.
`thinkpad_acpi does not load.`