vmatare/thinkfan #54: libboost requirement?

clickworkorange, 2018-06-04:
Trying to compile this but it fails when not finding libboost: 
````
In file included from /usr/include/yaml-cpp/node/node.h:16:0,
                 from /usr/include/yaml-cpp/yaml.h:16,
                 from /home/ola/Sources/thinkfan/src/yamlconfig.h:9,
                 from /home/ola/Sources/thinkfan/src/config.cpp:31:
/usr/include/yaml-cpp/node/ptr.h:11:32: fatal error: boost/shared_ptr.hpp: No such file or directory
compilation terminated.
````
However: 
````
apt-get install libboost-all-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libboost-atomic-dev libboost-atomic1.58-dev libboost-atomic1.58.0 libboost-chrono-dev libboost-chrono1.58-dev libboost-chrono1.58.0 libboost-context-dev
  libboost-context1.58-dev libboost-context1.58.0 libboost-coroutine-dev libboost-coroutine1.58-dev libboost-coroutine1.58.0 libboost-date-time-dev libboost-date-time1.58-dev
  libboost-dev libboost-exception-dev libboost-exception1.58-dev libboost-filesystem-dev libboost-filesystem1.58-dev libboost-graph-dev libboost-graph-parallel-dev
  libboost-graph-parallel1.58-dev libboost-graph-parallel1.58.0 libboost-graph1.58-dev libboost-graph1.58.0 libboost-iostreams-dev libboost-iostreams1.58-dev libboost-locale-dev
  libboost-locale1.58-dev libboost-log-dev libboost-log1.58-dev libboost-log1.58.0 libboost-math-dev libboost-math1.58-dev libboost-math1.58.0 libboost-mpi-dev
  libboost-mpi-python-dev libboost-mpi-python1.58-dev libboost-mpi-python1.58.0 libboost-mpi1.58-dev libboost-mpi1.58.0 libboost-program-options-dev
  libboost-program-options1.58-dev libboost-program-options1.58.0 libboost-python-dev libboost-python1.58-dev libboost-python1.58.0 libboost-random-dev libboost-random1.58-dev
  libboost-random1.58.0 libboost-regex-dev libboost-regex1.58-dev libboost-serialization-dev libboost-serialization1.58-dev libboost-serialization1.58.0 libboost-signals-dev
  libboost-signals1.58-dev libboost-signals1.58.0 libboost-system-dev libboost-system1.58-dev libboost-test-dev libboost-test1.58-dev libboost-test1.58.0 libboost-thread-dev
  libboost-thread1.58-dev libboost-timer-dev libboost-timer1.58-dev libboost-timer1.58.0 libboost-tools-dev libboost-wave-dev libboost-wave1.58-dev libboost-wave1.58.0
  libboost1.58-dev libboost1.58-tools-dev libhwloc-dev libhwloc-plugins libhwloc5 libibverbs-dev libibverbs1 libltdl-dev libnuma-dev libopenmpi-dev libopenmpi1.10 libtool
  mpi-default-bin mpi-default-dev ocl-icd-libopencl1 openmpi-bin openmpi-common
Suggested packages:
  libboost-doc graphviz libboost1.58-doc gccxml libmpfrc++-dev libntl-dev xsltproc doxygen docbook-xsl default-jdk fop libhwloc-contrib-plugins libtool-doc opennmpi-doc autoconf
  automaken gfortran | fortran95-compiler gcj-jdk opencl-icd gfortran openmpi-checkpoint
The following NEW packages will be installed
  libboost-all-dev libboost-atomic-dev libboost-atomic1.58-dev libboost-atomic1.58.0 libboost-chrono-dev libboost-chrono1.58-dev libboost-chrono1.58.0 libboost-context-dev
  libboost-context1.58-dev libboost-context1.58.0 libboost-coroutine-dev libboost-coroutine1.58-dev libboost-coroutine1.58.0 libboost-date-time-dev libboost-date-time1.58-dev
  libboost-dev libboost-exception-dev libboost-exception1.58-dev libboost-filesystem-dev libboost-filesystem1.58-dev libboost-graph-dev libboost-graph-parallel-dev
  libboost-graph-parallel1.58-dev libboost-graph-parallel1.58.0 libboost-graph1.58-dev libboost-graph1.58.0 libboost-iostreams-dev libboost-iostreams1.58-dev libboost-locale-dev
  libboost-locale1.58-dev libboost-log-dev libboost-log1.58-dev libboost-log1.58.0 libboost-math-dev libboost-math1.58-dev libboost-math1.58.0 libboost-mpi-dev
  libboost-mpi-python-dev libboost-mpi-python1.58-dev libboost-mpi-python1.58.0 libboost-mpi1.58-dev libboost-mpi1.58.0 libboost-program-options-dev
  libboost-program-options1.58-dev libboost-program-options1.58.0 libboost-python-dev libboost-python1.58-dev libboost-python1.58.0 libboost-random-dev libboost-random1.58-dev
  libboost-random1.58.0 libboost-regex-dev libboost-regex1.58-dev libboost-serialization-dev libboost-serialization1.58-dev libboost-serialization1.58.0 libboost-signals-dev
  libboost-signals1.58-dev libboost-signals1.58.0 libboost-system-dev libboost-system1.58-dev libboost-test-dev libboost-test1.58-dev libboost-test1.58.0 libboost-thread-dev
  libboost-thread1.58-dev libboost-timer-dev libboost-timer1.58-dev libboost-timer1.58.0 libboost-tools-dev libboost-wave-dev libboost-wave1.58-dev libboost-wave1.58.0
  libboost1.58-dev libboost1.58-tools-dev libhwloc-dev libhwloc-plugins libhwloc5 libibverbs-dev libibverbs1 libltdl-dev libnuma-dev libopenmpi-dev libopenmpi1.10 libtool
  mpi-default-bin mpi-default-dev ocl-icd-libopencl1 openmpi-bin openmpi-common
0 to upgrade, 90 to newly install, 0 to remove and 2 not to upgrade.
Need to get 16.6 MB of archives.
After this operation, 189 MB of additional disk space will be used.
Do you want to continue? [Y/n] HELL NO!
Abort.
````
A tad excessive, _for fan control_, isn't it? Also, this dependency isn't listed in the readme.

clickworkorange, 2018-06-04:
Never mind; desperate to get the fan to behave on my X200 so I installed the ~150Mb worth of dependencies to get this to build (i.e. cmake-curses, libboost-dev, etc). But it does rankle my minimalist sensibilities that such a small program has such big dependencies just for building. I also couldn't avoid installing libyaml-cpp even with `-D USE_YAML:BOOL=False` (`make` resulted in `fatal error: yaml-cpp/yaml.h: No such file or directory`). This all said, it does seem to be working! :)

vmatare, 2018-06-04:
Well, if you look closely, boost is actually not a direct dependency of thinkfan, but of the libyamlcpp you have installed. Unfortunately, certain distros still ship some ancient versions of libyamlcpp, and in your case it's so old that it still uses `boost::shared_ptr`.

That said, thinkfan should of course compile fine with `-D USE_YAML:BOOL=False`, so I'm reopening this to remind myself to look into that.

closed 2018-07-16 by vmatare