vmatare/thinkfan #56: fan stops for a moment after every timesleep (lenovo L480)

DEvil0000, 2018-07-26:
after every timesleep the fan is getting stopped/slowing down for a moment even if there is no change in level.

vmatare, 2018-07-26:
Are you using depulsing mode (the `-p` option)? Because without it, if there is no change to the fan level, thinkfan does absolutely nothing to the fan. So without `-p` it's highly unlikely that thinkfan is causing that.

DEvil0000, 2018-07-26:
well it is happening and exacly on sleeptime interval. if i change the sleeptime the fan slowdown is also changing to the same interval.no -p option is used. i used -n only, -n -s 1 and -n -b 1 -s 1 for testing.running ubuntu18.04 on the L480.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/vmatare/thinkfan/issues/56#issuecomment-408059237",
"url": "https://github.com/vmatare/thinkfan/issues/56#issuecomment-408059237",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [vmatare/thinkfan] fan stops for a moment after every timesleep (lenovo L480) (#56)",
"sections": [
{
"text": "",
"activityTitle": "**Victor Mataré**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@vmatare",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"vmatare/thinkfan\",\n\"issueId\": 56,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"vmatare/thinkfan\",\n\"issueId\": 56\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/vmatare/thinkfan/issues/56#issuecomment-408059237"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 361113043\n}"
}
],
"themeColor": "26292E"
}
]

vmatare, 2018-07-31:
Could you please post the config file you use and which thinkfan revision, i.e. `git show HEAD`...?

DEvil0000, 2018-08-01:
seems like the issue was with a older version packaged in ubuntu 18.04. package version=0.9.3-2, thinkfan binary says 0.9.1 in the help.
however it works as expected with master.
thanks

DEvil0000, 2018-08-01:
with the master version the fan is now slowing down when the level changes downwards. For example if the fan level goes from 7 to 5 the fan stops for a moment before it spins up again to the speed of 5 which is of course slower then 7.

acarapetis, 2018-08-29:
Sadly, I think this is a hardware/firmware issue - I notice the same thing on my E480, but it's not thinkfan's problem: you can observe the same behaviour by manually adjusting the fan speed using `/proc/acpi/ibm/fan`, or even using tpfancontrol on Windows.

vmatare, 2020-04-08:
Closing since this has been silent for a while and it's (probably) not fixable from within thinkfan.

closed 2020-04-08 by vmatare