vmatare/thinkfan #57: Spam in dmesg after commit d3951070f95ffe1ec5e4505c58387edb63206b8a

susanwl, 2018-08-06:
Hi,

Apparently for me commit `d3951070f95ffe1ec5e4505c58387edb63206b8a` makes `thinkpad_acpi` driver to constantly spam into dmesg(every 2 minutes) like:
```
[Aug 6 05:14] thinkpad_acpi: fan watchdog: enabling fan
[Aug 6 05:16] thinkpad_acpi: fan watchdog: enabling fan
[Aug 6 05:18] thinkpad_acpi: fan watchdog: enabling fan
[Aug 6 05:20] thinkpad_acpi: fan watchdog: enabling fan
[Aug 6 05:22] thinkpad_acpi: fan watchdog: enabling fan
```

Also I am a bit sceptical of the change brought by that commit:
```
        }
-       else if (last_watchdog_ping_ + watchdog_ + sleeptime >= std::chrono::system_clock::now())
+       else if (last_watchdog_ping_ + watchdog_ + sleeptime <= std::chrono::system_clock::now())
         set_speed(level);
 }
```
, since when `set_speed` is called `last_watchdog_ping_` is set to the current time `src/drivers.cpp:136`:
```
last_watchdog_ping_ = std::chrono::system_clock::now()
```
so the condition
```
else if (last_watchdog_ping_ + watchdog_ + sleeptime <= std::chrono::system_clock::now())
```
should always be false, unless I am missing something.

vmatare, 2018-08-24:
Good catch, looks like since commit d395107 we're always pinging the watchdog `sleeptime` seconds too late. However I think `min(watchdog_, sleeptime)` isn't the best approach. If I'm not overlooking anything, the watchdog is currently always set to 120 seconds, so we want to _subtract_ the sleeptime to make sure we ping now if we know that it's going to trigger during the next sleep period.

vmatare, 2018-08-24:
It would be great if you could test again with commit fd94d53263863d475f7284b1be2df2197c72920e.

closed 2018-08-28 by vmatare