vmatare/thinkfan #65: Errors out on nonerror condition?

StyXman, 2019-05-23:


vmatare, 2019-05-24:
You're using the wrong keyword. `tp_thermal` is for the stuff in `/proc/acpi/ibm/thermal`, which has a completely different format. For temperature sources in sysfs you have to use the `hwmon` keyword.

vmatare, 2019-05-24:
However the error message is indeed misleading, so that is a bug indeed.

vmatare, 2020-04-08:
Huh. If I use the `tp_thermal` keyword with an hwmon sensor, I get:
```
ERROR: /sys/class/hwmon/hwmon0/temp1_input: Unknown file format.
```
Which is a (more) correct error message for this case.

StyXman, 2020-04-09:
Hi @vmatare . That was a year ago. [According to Debian's archive](https://snapshot.debian.org/archive/debian/20190523T025328Z/pool/main/t/thinkfan/), it should have been 0.9.3 or so. I don't even have `thinkfan` installed anymore. If the current version shows a message like you shown, then you can consider this one closed.

vmatare, 2020-04-09:
Great, thanks for the feedback!

closed 2020-04-09 by vmatare