vmatare/thinkfan #67: thinkpad_acpi module doesn't load at boot

zjeffer, 2019-06-09:
I added `thinkpad_acpi fan_control=1` to `GRUB_CMDLINE_LINUX_DEFAULT` in /etc/default/grub.
I also added `options thinkpad_acpi fan_control=1` to /etc/modprobe.d/thinkfan.conf, just to be sure.

Neither of these load the required module. I have to manually do `sudo modprobe thinkpad_acpi` to load the module. 

Because it doesn't load at boot, thinkfan.service crashes, saying it can't find the /proc/acpi/ibm/fan, because the ibm directory doesn't exist if the module doesn't load.

How do I load the module at boot?

zjeffer, 2019-06-09:
Adding a file called `thinkpad.conf` to /etc/modules-load.d/ with the line `thinkpad_acpi` fixed it. I don't know why the other things I tried don't make it load the module at boot.

closed 2019-06-09 by zjeffer

vmatare, 2019-06-10:
Well, the other two things you did are actually equivalent and they both only **set the parameter** on the `thinkpad_acpi` module, but they do nothing to load it. Loading is precisely what `/etc/modules-load.d/` is for, just as the name suggests. That's how it is designed in every modern Linux distribution.

TUX-A1CUA0, 2024-11-16:
> Adding a file called `thinkpad.conf` to /etc/modules-load.d/ with the line `thinkpad_acpi` fixed it. I don't know why the other things I tried don't make it load the module at boot.

does it still mork

zjeffer, 2024-11-16:
This issue is 5 years old, I have another laptop now and stopped using this. Why not try it out yourself and let us know? ;)

Like @vmatare says, you simply have to load the kernel module at boot and then it should work. If it doesn't, you should create a new issue and describe exactly what doesn't work.