vmatare/thinkfan #68: libyaml-cpp or yaml-cpp

pooyataher, 2019-08-19:
In the README it says that the program needs libyaml-cpp to be configured using a YAML configuration file.  Does it mean this `[libyaml-cpp](https://github.com/jasonjei/libyaml-cpp)` or `[yaml-cpp](https://github.com/jbeder/yaml-cpp)`?
The former doesn't seem to be a well maintained program as there is neither a Makefile nor can it be installed.

pooyataher, 2019-08-19:
From an old [comment](https://github.com/vmatare/thinkfan/issues/42#issuecomment-336539089) of yours, it seems that you mean [ymal-cpp](https://github.com/jbeder/yaml-cpp) by libyaml-cpp.
Why not correct the README to prevent any confusion?
By the way, I couldn't find any Ubuntu package repository (APT) for libyaml-cpp or yaml-cpp.  That is why I tried to install from the source code.  Hence the confusion.
In the end, I really appreciate the code for `thinkfan`!

vmatare, 2019-08-19:
Huh. That is strange. I specifally wrote `libyaml-cpp` (resp. `libyaml-cpp-dev`) because that is precisely what the package is called in Ubuntu: https://packages.ubuntu.com/de/xenial/libyaml-cpp-dev
Seems to exists in 18.04 as well: https://packages.ubuntu.com/de/bionic/libyaml-cpp-dev

vmatare, 2019-08-19:
Of course it refers to https://github.com/jbeder/yaml-cpp, never mind that other thing that I've never heard about. Generally one tries to depend on things that are actually shipped by distros ;-)

pooyataher, 2019-08-19:
Yes, you are right `libyaml-cpp-dev` is actually an Ubuntu package.  Thank you!

closed 2019-08-20 by pooyataher