vmatare/thinkfan #70: Build error: missing init & init_fans methods

nightuser, 2019-10-02:
Hey!

I tried to build the recent git master and I got the following error:

```
$ make
[ 12%] Building CXX object CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o
/home/nightuser/.cache/aurutils/sync/thinkfan-git/aaa/src/thinkfan.cpp: In function ‘int main(int, char**)’:
/home/nightuser/.cache/aurutils/sync/thinkfan-git/aaa/src/thinkfan.cpp:422:16: error: ‘class thinkfan::TemperatureState’ has no member named ‘init’
  422 |     temp_state.init();
      |                ^~~~
/home/nightuser/.cache/aurutils/sync/thinkfan-git/aaa/src/thinkfan.cpp:423:15: error: ‘const class thinkfan::Config’ has no member named ‘init_fans’
  423 |     test_cfg->init_fans();
      |               ^~~~~~~~~
/home/nightuser/.cache/aurutils/sync/thinkfan-git/aaa/src/thinkfan.cpp:456:12: error: ‘const class thinkfan::Config’ has no member named ‘init_fans’
  456 |    config->init_fans();
      |            ^~~~~~~~~
make[2]: *** [CMakeFiles/thinkfan.dir/build.make:63: CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/thinkfan.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Indeed, there are no such methods in these classes.

vmatare, 2019-10-02:
Thanks for the report, I just wanted to see if anyone is paying attention :-P
Should be fixed in the master now. It's supposed to become a minor release before the big `optional-sensors` and `multifan` features, so I highly appreciate all feedback.

vmatare, 2019-10-04:
OK, so since it's fixed for me now, I'm assuming this is done ;-)

closed 2019-10-04 by vmatare

nightuser, 2019-10-04:
@vmatare Thanks! Late-night coding is not the best idea for sure. At least, pushing it to the repo.

vmatare, 2019-10-04:
Well, the tested & stable stuff goes into the releases, so the `master` on github here should not be considered stable by any means.

If you look closely you'll notice that I don't have a lot of testing infrastructure in place here. By the nature of what thinkfan does, and the way e.g. the Linux sysfs is structured and in the way thinkfan interacts with your hardware, testing everything _properly_ would require a huge amount of additional tooling.

That is why I have to rely on the feedback from users like you, and why I push to the `master` as soon as I want something to be tested out in the wild. On some hidden development branch that just wouldn't happen. Naturally I'll try to avoid bothering people with boring build failures, but once in a while it's also good for me to see that I do actually get reports when I've pushed crap ;-)