vmatare/thinkfan #72: Hungs with wrong sensor path

diggit, 2019-10-04:
There were probably some changes in kernel and sensors moved to different places in hwmon hierarchy. Sensor I chose to control thinkfan was in `/sys/class/hwmon/hwmon2/`. Now, `hwmon2` is empty and sensor moved to `/sys/class/hwmon/hwmon4/`. Problem is, that thinkfan with wrong config hung at startup and only possible way to terminate it is by sending SIGKILL(9).

Expected result reporting error into log and exiting with error or running all fans at max RPM.

Happens with thinkfan release 1.0.2 and same result was with master branch ~3 days ago.

vmatare, 2019-10-04:
Hm, cannot reproduce. Can you please post the exact config you're using?

diggit, 2019-10-04:
Yep, here it is:
```
sensors:
  - hwmon: /sys/class/hwmon/hwmon2
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

#
# Complex level config
#
levels:
  - speed: level auto
    upper_limit: [95]

  - speed: level disengaged
    lower_limit: [80]
    #upper_limit: [60, 60, 60]

#
# Simple level config
# meaning: [FANSPEED, LOWER_LIMIT, UPPER_LIMIT]
#
#levels:
#  - [0, 0, 50]
#  - [2, 45, 60]
```
`hwmon2` exists, but there is no `temp1_input` which moved to `hwmon4`.

vmatare, 2019-10-04:
> hwmon2 exists, but there is no temp1_input which moved to hwmon4.

Oh, I think this bit is the crucial detail ;-)

vmatare, 2019-10-04:
Still can't repro. Are you sure you're using the latest `master`? What's the output of of `git show -q HEAD`?

vmatare, 2019-10-04:
The issue is present in the 1.0.2 release though, but it should be fixed by [7a14999](https://github.com/vmatare/thinkfan/commit/7a1499956b572a1692b267e424ec31b32e0d16b2)

diggit, 2019-10-04:
Yeah, current master works! My build from master was probably before mentioned commit.
```
Oct 04 15:07:49 dgtsnb thinkfan[48276]: ERROR: Could not find an `hwmon*' directory or `temp*_input' file in /sys/class/hwmon/hwmon2/device/device.
Oct 04 15:07:49 dgtsnb systemd[1]: thinkfan.service: Control process exited, code=exited, status=1/FAILURE
Oct 04 15:07:49 dgtsnb systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Oct 04 15:07:49 dgtsnb systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
```

closed 2019-10-04 by diggit