vmatare/thinkfan #74: Thinkfan doesn't start anymore (T 400)

mariobl, 2019-10-12:
Hello,

on my Thinkpad T 400 is doesn't start anymore.

[root@t400 mariobl]# systemctl status thinkfan.service
● thinkfan.service - simple and lightweight fan control program
   Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
  Drop-In: /etc/systemd/system/thinkfan.service.d
           └─override.conf
   Active: failed (Result: exit-code) since Sat 2019-10-12 10:30:08 CEST; 5min ago
  Process: 1931 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS (code=exited, status=127)

Okt 12 10:30:08 t400 systemd[1]: Starting simple and lightweight fan control program...
Okt 12 10:30:08 t400 thinkfan[1931]: /usr/bin/thinkfan: symbol lookup error: /usr/bin/thinkfan: undefined symbol: _ZN4YAML6detail9node_data12empty_scalarB5cxx11E
Okt 12 10:30:08 t400 systemd[1]: thinkfan.service: Control process exited, code=exited, status=127/n/a
Okt 12 10:30:08 t400 systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Okt 12 10:30:08 t400 systemd[1]: Failed to start simple and lightweight fan control program.

Output of hwinfo:
https://pastebin.com/raw/quSv4kQm

Admittedly, my Thinkpad is quite old, and sometimes the BIOS shows an error message "Fan error" at startup... But even in case the error message doesn't appear, thinkfan doesn't start. Attempting to start it manually, without systemctl:

[root@t400 mariobl]# thinkfan
thinkfan: symbol lookup error: thinkfan: undefined symbol: _ZN4YAML6detail9node_data12empty_scalarB5cxx11E

vmatare, 2019-10-16:
Hi, this looks to me like you compiled thinkfan from the source code, but your `libyaml-cpp` has been updated since then. Have you tried recompiling?

sync0, 2019-10-26:
I am having the exact same problem and `systemctl status thinkfan.service` shows the same error message. My thinkpad is an X230 (2013), so I guess something else is causing the problem.

mariobl, 2019-10-26:
For me, the problem is solved, @vmatare, thanks for the hint. This is an internal problem in my Archlinux distribution: Thinkfan is only available from the Arch User Respository and not part of the official package pool. That's why nobody cares about such changes in libraries which are needed for thinkfan.

vmatare, 2019-10-27:
Well, then your package manager should know that `thinkfan` depends on `libyaml-cpp` and rebuild it automatically whenever `libyaml-cpp` is updated. If it does not, then there's either a dependency missing from the package definition or it's a bug in the package manager.

vmatare, 2019-10-27:
This has nothing to do with anybody's hardware btw. The error says that the function `YAML::detail::node_data::empty_scalar[abi:cxx11]` which existed (in libyaml-cpp) when thinkfan was built does not exist any more. That's something that can only happen if `libyaml-cpp` has been updated in the meantime.

bssb, 2019-10-30:
I maintain the thinkfan AUR package and the PKGBUILD does correctly define yaml-cpp as a dependency. If your AUR helper is unable to detect when dependencies are updated, then you need to recompile thinkfan yourself. This issue should be closed.

mariobl, 2019-10-30:
Indeed, for me the problem has been solved. But look at the comment of @sync0, he seems to have the same problem, at least the same error message.

bssb, 2019-10-30:
Yeah, my comment about needing to recompile was mainly directed towards him, confirming the solution already discussed here.

sync0, 2019-10-31:
Thank you for the helpful comments. In the end, I was able to get thinkfan back to work after a manual install. I use `pacaur` to take care of AUR packages, and it seems that `pacaur` had problems with the package `auracle`, which it requires. After manually installing `auracle` I got `pacaur` back to work, and it seems that my problem with `thinkfan` was that `pacaur` was not able to deal with it.

mariobl, 2019-11-01:
OK, then let's close this issue.

closed 2019-11-01 by mariobl