vmatare/thinkfan #75: “level auto” does not work with YAML config

Flupp, 2019-10-15:
Using the following YAML config:
```YAML
sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [1, 2, 3, 4, 5]

fans:
  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon
    indices: [1]

levels:
  - ["level 0"   ,  0,    60]
  - ["level 1"   , 45,    75]
  - ["level auto", 70, 32767]
```
… I get the following error for `thinkfan -n`:
```
ERROR: You're using a PWM fan, but your highest fan level is only -2147483648. Enable DANGEROUS mode if you're really sure you never need to max out your fan.
```
… and for `thinkfan -n -D` while raising the temperature I get:
```
ERROR: You're using a PWM fan, but your highest fan level is only -2147483648. Enable DANGEROUS mode if you're really sure you never need to max out your fan.
ERROR: You're using a PWM fan, but your highest fan level is only -2147483648. Enable DANGEROUS mode if you're really sure you never need to max out your fan.
Temperatures(bias): 41(0), 39(0), 40(0), 40(0), 40(0) -> level 0
Temperatures(bias): 67(39), 47(10), 45(7), 67(39), 46(9) -> level auto
ERROR: set_speed: Writing "-2147483648" to /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon4/pwm1: Invalid argument
```
In both cases, thinkfan returns with error value 1.

vmatare, 2019-10-15:
Hi, it has nothing to do with the YAML config. It is the fact that you're trying to use a `hwmon` fan with thinkpad fan levels. There is no `level auto` in a hwmon PWM fan. Please read the config man page `thinkpad.conf(5)`, section FANS. Although it does not describe the YAML config syntax, what is said there about PWM fans vs. `thinkpad_acpi` fans applies generally.

At least that is the way it's currently implemented. That said, your intuition isn't entirely wrong, because with PWM we can just as well turn off userspace fan control, thereby giving it back to the kernel or the hardware. However that is not currently implemented, so I'm turning this into a feature request.

vmatare, 2019-10-16:
Also the error message isn't exactly helpful here ;-)

vmatare, 2019-10-16:
BTW, it looks like you're using a thinkpad, so what you're probably meaning to do is this:
```YAML
fans:
  - tpacpi: /proc/acpi/ibm/thermal
```

Flupp, 2019-10-16:
Thank you!
```YAML
fans:
  - tpacpi: /proc/acpi/ibm/fan
```
This is working.