vmatare/thinkfan #77: Report unusual sensor readings

mmtj, 2019-10-31:
Hi,
I couldn't figure out why fan isn't spinning on my thinkpad x201 and after quick search I found issues #45 and #75. These pointed me to the right direction - which thermal sensor am I using?

`/proc/acpi/ibm/thermal` is used by default, however in my case, it gave unusual values.
```
# cat /proc/acpi/ibm/thermal
temperatures:	-128 0 0 0 0 0 0 0
```

After little bit of digging, I figured out that correct thermal sensor is `hwmon`, which gives expected temperature values (compared to `sensors` utility output) and set `hwmon` in config. Fan now spins as expected.

However debugging could be much easier if there were some sort of warning that could point user to possible cause. Something like:
```
Getting unusual values from thermal sensor (<value>).
```

Also, I am not sure whether it is a bug or intentional behaviour: When thinkfan reads negative temperatures, it will stop fan.

vmatare, 2019-10-31:
Well, the important question is: What is an unusual value?
Because -128 is definitely not unusual. In `/proc/acpi/ibm/thermal` it's a reserved value used to indicate that a certain sensor is not in use on a particular platform. Consequentially, the sensor should be ignored, and that is just what happens if we use the -128 °C in the speed calculation as-is.

What looks more unusual to me is that all the other temperatures are 0. However 0 in itself is also not an unusual value. The problem seems to be that they're all 0 and don't seem to change (?). So what would be required here is a noise estimation, and only if the noise across all sensors is 0, we can assume something is wrong. Sounds like an interesting feature, but I'm not sure if we want the added complexity.

mmtj, 2019-10-31:
Well, it seems more complex that it looked at first and I agree that there should be some heuristic used to determine how unusual these values are. Besides marking negative values as error could lead to false positives in case of sub-zero temperatures. 

Where did you get reference for reserved values? I got impression that all negative values means inaccessible sensor, but can't confirm it as kernel thinkpad_acpi docs doesn't mention it. Or perhaps I just misread something.