vmatare/thinkfan #79: ERROR: /etc/thinkfan.conf: Invalid entryInvalid entry.

devsnek, 2019-11-12:
When I run thinkfan with this config:

```
#
# IBM/Lenovo Thinkpads (thinkpad_acpi, /proc/acpi/ibm)
# ====================================================
#
# IMPORTANT:
#
# To keep your HD from overheating, you have to specify a correction value for
# the sensor that has the HD's temperature. You need to do this because
# thinkfan uses only the highest temperature it can find in the system, and
# that'll most likely never be your HD, as most HDs are already out of spec
# when they reach 55 °C.
# Correction values are applied from left to right in the same order as the
# temperatures are read from the file.
#
# For example:
# tp_thermal /proc/acpi/ibm/thermal (0, 0, 10)
# will add a fixed value of 10 °C the 3rd value read from that file. Check out
# http://www.thinkwiki.org/wiki/Thermal_Sensors to find out how much you may
# want to add to certain temperatures.

#  Syntax:
#  (LEVEL, LOW, HIGH)
#  LEVEL is the fan level to use (0-7 with thinkpad_acpi)
#  LOW is the temperature at which to step down to the previous level
#  HIGH is the temperature at which to step up to the next level
#  All numbers are integers.
#

tp_thermal /proc/acpi/ibm/thermal
tp_fan /proc/acpi/ibm/fan

(0, 0,  55)
(1, 48, 60)
(2, 50, 61)
(3, 52, 63)
(4, 56, 65)
(5, 59, 66)
(7, 63, 32767)
```

```
$ sudo thinkfan

ERROR: /etc/thinkfan.conf: Invalid entryInvalid entry.
```

0xA1B2, 2019-11-20:
Post the result of:

`cat /proc/acpi/ibm/fan`

devsnek, 2019-11-20:
```
status:		enabled
speed:		2895
level:		auto
commands:	level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:	enable, disable
commands:	watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))
```

0xA1B2, 2019-11-20:
That looks good.

Now comment out these two lines and try again:

#tp_thermal /proc/acpi/ibm/thermal
#tp_fan /proc/acpi/ibm/fan

devsnek, 2019-12-23:
same error.

devsnek, 2019-12-23:
even an empty file gives the error.

devsnek, 2019-12-23:
i gave up and created https://github.com/devsnek/thonkfan

vmatare, 2020-04-04:
I can't reproduce this with the current `master` branch. If anyone is still experiencing this, please run cmake with `-DDISABLE_EXCEPTION_CATCHING=ON` and rebuild. Then thinkfan should output a backtrace.

vmatare, 2020-04-16:
This looks like for some reason, thinkfan is trying to interpret a legacy config as YAML format. Normally, if thinkfan was compiled with `USE_YAML=ON`, it will first try to read the config in YAML format, and if that fails, it will try the legacy format. But the error message looks like one from the YAML parser, as if the legacy format had never been tried. So I cannot investigate this without either a backtrace or more specific information on the thinkfan version and the exact setup used.

closed 2020-04-16 by vmatare