vmatare/thinkfan #81: undefined symbol

cbrace, 2019-11-18:
I installed the thinkfan v1.0.2 rpm on a Thinkpad X220 running Fedora v31. When I start the program, I get an error message:

```$ sudo thinkfan -n
thinkfan: symbol lookup error: thinkfan: undefined symbol: _ZN4YAML6detail9node_data12empty_scalarB5cxx11E
```
Contents of /etc/thinkfan.conf below. Please let me know if additional info is required.
```
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon4/temp3_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon4/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon4/temp2_input
hwmon /sys/devices/virtual/thermal/thermal_zone0/hwmon1/temp1_input

(0,     0,	55)
(1,     48,     60)
(2,     50,     61)
(3,     52,     63)
(4,     56,     65)
(5,     59,     66)
(7,     63,     32767)`
```

vmatare, 2019-11-18:
You'll have to report that to the maintainer of the Fedora package, that's something that they need to take care of.

vmatare, 2019-11-18:
Another one with broken linkage, now Fedora
Arch: #74

rathann, 2019-12-03:
This was actually due to `yaml-cpp` update which changed ABI without changing `SONAME`. Downstream bug: https://bugzilla.redhat.com/show_bug.cgi?id=1773357 is fixed already. You might as well close this one.

closed 2019-12-03 by vmatare