vmatare/thinkfan #92: Can't figure out how to add correction values or indices in the new yaml format

thiswillbeyourgithub, 2020-04-18:
Hi,

After some overheating issue caused by a hwmon number that chaned at my last reboot I finally updated my thinkfan to the last version (1.1). I spent a good part of my afternoon struggling to understand how to specify the correction values, I keep getting 
```ERROR: SensorDriver: Initializing fan control in /sys/devices/virtual/hwmon/hwmon0/temp1_input [3]: No such file or directory```

Right know my config contains this line for example : 
>    - hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input [3]  
      # name: acpiz

I feel like I have tried every combination of parentheses or brackets but I can't figure out how to make it work. I tried [3], (3) [(3)] etc. Also strict indentation and spaces seem to matter. Of course removing the correction values work so the path is correct.

Also I don't understand how indices work, specifically : sometimes my thinkpad T480s loads `hwmon2` as `hwmon3` so the old thinkfan would not launch. Can you point me to the general direction ? I am trying to work out how to use the new indices feature.

Thanks! Also this utility really rocks so thanks :)

vmatare, 2020-04-19:
Are you using the current `master` branch? Because I have just updated the `thinkfan.conf` manpage to cover the YAML format. Take a look and let me know if you're missing anything ;-)

vmatare, 2020-04-19:
To make it short: You need to have a separate entry like this:
```yaml
sensors:
  - hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input
    correction: [3]
```

thiswillbeyourgithub, 2020-04-19:
Thanks a lot for the quick answer!
I indeed got this working.

Also, I don't know if it's up to you but the error message is not very intuitive when indicating that indentation is incorrect in the YAML fild. I have no knowledge of YAML whatsoever and was surprised to see that had I not been using vim, I would have found it hard to figure out that one line was one extra space to the right.

Thanks a lot :)!

thiswillbeyourgithub, 2020-04-19:
Actually, I'm sorry but after playing around with the YAML file I still have trouble with optional values. man thinkfan and man thinkfan.conf are not helping.

I have this path : 
```    - hwmon: /sys/devices/virtual/hwmon/hwmon2/temp1_input```
Apparently, hwmon2 can also be called hwmon1 in some cases, so the path can vary. That's what was causing me so many issues over the years with the previous version.

I thought this problem was adressed with the "optional" feature, yet I can't make this work : 

```
sensors:
    # - hwmon: /sys/class/graphics/fb0/device/hwmon
    # indices: [1, 3]

    # - hwmon: /sys/devices/virtual/thermal
    #  name: pch_skylake
    #  indices: [1]
  
    #- tpacpi: /proc/acpi/ibm/thermal

    - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
      # name: id0
    - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
      # name: core 0
    - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
      # name: core 1
    - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp4_input
      # name: core 2
    - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp5_input
      # name: core 3
    - hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input 
      correction: [3]
      # name: acpiz
    - hwmon: /sys/devices/virtual/hwmon/hwmon1/temp1_input
      correction: [7]
      optional: true
      # name: skylake can be hwmon1 or 2
    - hwmon: /sys/devices/virtual/hwmon/hwmon2/temp1_input
      correction: [7]
      optional: true
      # name: skylake can be hwmon1 or 2
    - hwmon: /sys/devices/virtual/hwmon/hwmon4/temp1_input
      correction: [7]
      # name: wifi

fans:
    # - hwmon: /sys/class/graphics/fb0/device/hwmon
    # indices: [1]
  - tpacpi: /proc/acpi/ibm/fan

```

Would you mind helping me once more ? I am sorry, I feel bad for reopenninng this :/

vmatare, 2020-04-19:
It's totally fine, don't worry.

Changing sensor paths have nothing to do with the "optional sensors" feature. An optional sensor is one that may disappear while thinkfan is running. Something that would normally be an error and cause thinkfan to exit.

What you're looking for is index- or name-based search. You can remove the `optional: true` there, because that won't help you. And then you need to read the `thinkfan.conf` manpage. Pay attention to the **Sensor Syntax** section and in particular the explanations for the **hwmon-path**, **hwmon-name** and **index-list** values.

thiswillbeyourgithub, 2020-04-21:
Ohhhh

Turns out I compiled it with `RelWithDebInfo` as type, I don't know if it's the cause but it didn't install the new thinkfan.conf man page, so it was impossible to understand. I reinstalled with  `Release` and now I can indeed access the syntax part you're refering to. And indeed saw warnings about the intendation while writing in YAML format.

Thanks a lot for the kind help :)

But. I tried and tried and still get confusing errors. I commented most of the `thinkfan.conf.yaml` to get as little variables as possible. To understand the syntax I'm focusing on hwmon4, here's my conf file : 

```
sensors: 
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
    # name: those are the temperature of the cores
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
    # name: core0
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
    # name: core1
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp4_input
    # name: core2
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp5_input
    # name: core3

    #- hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input 
    #name: acpitz
    #correction: [3]

    #- hwmon: /sys/devices/virtual/hwmon/hwmon2/temp1_input
    #name: pch_skylake
    #    indices: [1, 2]
    #correction: [7]

  - hwmon: /sys/devices/virtual
    name: iwlwifi
    indices: [ 4 ]
    correction: [7]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
- [0, 0,  55]
- [1, 52, 68]
- [2, 60, 70]
- [3, 60, 75]
- [4, 68, 72]
- [5, 65, 80]
- [6, 72, 76] 
- [7, 73, 32767]
```

Here's the output of pwd, the content of "name" and "temp1_input" files :  
```
[13:58]hwmon4 pwd
/sys/devices/virtual/hwmon/hwmon4
[14:05]hwmon4 cat name temp1_input 
iwlwifi
38000
```

When I'm trying to run thinkfan I get this : 
```
[14:00]/etc sudo thinkfan -c thinkfan.conf.yaml
ERROR: Could not find an `hwmon*' directory or `temp*_input' file in /sys/devices/virtual/hwmon/hwmon4.
```

vmatare, 2020-04-21:
Don't worry about bugging anyone. The manpage is new and you're actually helping me improve it. So now I made an update to the paragraph that explains how to specify a _hwmon-path_ under **Values** in the **SENSOR & FAN DRIVERS** section. It would be great if you could give it another read and see if that helps you figure out what you need to do.

Sorry I'm going this slightly complicated way, but there's really no point in explaining it here, since I want to clarify these things once and for all, not just for the few persons who see this issue report ;-)

thiswillbeyourgithub, 2020-04-21:
Thanks :)

Okay I think I just figured it out and it works now. I put `indices: [1]` as there is only one interface and it seems okay.

Is declaring `indices` really necessary if only one is found ? I found that quite confusing.

If I may suggest, I think the man should add *real* examples instead of more general formulas.
For the scenarios in the thinkfan.conf man : 

1)  
```
   - hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input        # <- non ambiguous path, no indices nor name needed
      correction: [3] # <- optionnal correction values 
```

2)
```
 - hwmon:  /sys/devices/platform/nct6775.2592      # <- path to one specific driver, only indices are needed
       correction: [7] 
       indices: [2, 3] # <- which interfaces of the driver to listen to 
```

3)
```
  - hwmon:  /sys/class/hwmon   # <- contains several sensors, indices and name are needed
     indices: [2, 3] # <- which interface to listen to 
     name: pch_skylake  # <- name of the specific sensors thinkfan should listen to
```

I think real scenarios like those should make their way into the man, or at least in the `example/thinkfan.conf.yaml`, as I think the comments are really insightful.

I will put my final conf file here in case someone else needs guidance :)

thiswillbeyourgithub, 2020-04-21:
As promised, here's my config file for my Thinkpad T480s : 
```
sensors: 

#those are the temperature of the cores
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
    #core0
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input
    #core1
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input
    #core2
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp4_input
    #core3
  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp5_input

    #acpitz
  - hwmon: /sys/devices/virtual/hwmon/hwmon0/temp1_input 
    correction: [3]

  - hwmon: /sys/devices/virtual/hwmon/
    correction: [7]
    name: pch_skylake
    indices: [1]

    #iwlwifi
  - hwmon: /sys/devices/virtual/hwmon/hwmon4/temp1_input
    correction: [7]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
- [0, 0,  55]
- [1, 52, 68]
- [2, 60, 70]
- [3, 60, 75]
- [4, 68, 72]
- [5, 65, 80]
- [6, 72, 76] 
- [7, 73, 32767]
```

vmatare, 2020-04-21:
I took another and noticed another possible misunderstanding regarding the meaning of the `indices:` entry. I've tried to clarify it in another update to the config manpage.

thiswillbeyourgithub, 2020-04-21:
Looks good to me. Thanks a lot :) Feel free to close this issue.

vmatare, 2020-04-23:
I'll close it when I've looked over the examples once more

vmatare, 2020-06-05:
We now have a big fat example config that should cover every aspect of the YAML config. The man page is intended as more of a reference, so I'll keep that more compact and abstract. I think in conjunction the two should really make things clear.

closed 2020-06-05 by vmatare