vmatare/thinkfan #93: YAML syntax for nvml sensor in tihinkfan.conf

kantor304, 2020-04-19:
Hello,

I am having difficulties getting my head around the yaml syntax for adding a nvml sensor in the thinkfan.conf.

In my case, thinkfan V1.2 was compiled with the default settings, so nvml should be active. Using one of the hwmon sensors thinkfan runs fine. However, adding a nvml sensor gives "Invalid sensor entry" or "Invalid syntax" depending on the attempted syntax.

`lspci | grep -i nvidia` 

results in

`01:00.0 VGA compatible controller: NVIDIA Corporation GF106GLM [Quadro 2000M] (rev a1)`

What would be the correct way of adding the GPU temperature to the sensor list in the yaml-format thinkfan.conf? An example would be very helpful.

Thanks in advance!

Jonas

vmatare, 2020-04-19:
Thanks for the report. I actually found a bunch of other bugs when trying to reproduce this one. Anyways, it should work as described in the manpage `thinkfan.conf (5)` now with the `nvml` keyword. That is, in your case:
```yaml
sensors:
  - nvml: 01:00.0
  - (etc...)
```

kantor304, 2020-04-19:
Thank you for the very fast response!

The entry

`  - nvml: 01:00.0`

under `sensors:` still gives "Invalid sensor entry". Let me know if there are any other things I should try.

vmatare, 2020-04-19:
That should definitely be fixed. Can you post the output of:
```bash
git show -q HEAD
```
and
```bash
objdump -TC `which thinkfan` | grep -i 'convert.*nvml'
```
If you haven't installed thinkfan into `$PATH`, then replace the `` `which thinkfan` `` with the path to your `thinkfan binary`.

If the commit you see from the first command is from Apr 19 17:24:17 2020 or later, and you do get an output from the second command, you should also post your entire config so I can try to reproduce the problem.

vmatare, 2020-04-19:
Oh and also, please always copy&paste the **entire** error message. Some detail in there that looks insignificant to you might actually help developers figure out the problem.

kantor304, 2020-04-20:
Hello,

Sure;

`git show -q HEAD`
`fatal: not a git repository (or any of the parent directories): .git`

Could this be since I downloaded the .zip file rather than the repository? The files in the .zip are dated 2020-04-15. Does that mean there is a later version available which I should use?

`objdump -TC `which thinkfan` | grep -i 'convert.*nvml'`
`0000000000043ae0  w   DF .text	00000000000003d0  Base        YAML::convert<YAML::wtf_ptr<thinkfan::NvmlSensorDriver> >::decode(YAML::Node const&, YAML::wtf_ptr<thinkfan::NvmlSensorDriver>&)`

I am not sure I understand your note about install path (beginner speaking here). Apart from `cmake -D CMAKE_BUILD_TYPE:STRING=Release` no flags or options were used for `make`, `build` or `build install`. No particular install path was specified.

For the record, the entire error message when using ` - nvml: 01:00.0` is:

`sudo thinkfan -n`
`ERROR: /etc/thinkfan.conf: Invalid sensor entry.
You have an ancient libyaml-cpp which can't give line numbers on errors. Please complain to your Linux distribution.`

If using `  - hwmon: /sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input ` thinkfan works fine.

I am using the yaml version which comes with Ububtu 18.04. Sorry for it being out of date. 

Again, thanks for the fast response!

vmatare, 2020-04-20:
Good, thanks for the feedback. So it's quite simple. You did everything right with compiling, it's just this:

> `git show -q HEAD`
> `fatal: not a git repository (or any of the parent directories): .git`
> 
> Could this be since I downloaded the .zip file rather than the repository? The files in the .zip are dated 2020-04-15. Does that mean there is a later version available which I should use?

Precisely. That .zip download is just the files at the time you're downloading it. If you want to keep up with development you should really use git. You do `git clone git@github.com:vmatare/thinkfan.git`, and after that you can simply update by running `git pull` in the `thinkfan` directory.

kantor304, 2020-04-21:
Problem solved!

After cloning, compiling and installing the current version from the repositories, thinkfan happily reads the GPU temperature defined in the thinkfan.conf as ` - nvml: 01:00.0`.

For the record:

`git show -q HEAD`

`commit 7fd32b2efb0ac49d7bb9f9ce13505c0a9ffb4f5a (HEAD -> master, origin/master, origin/HEAD)`
`Author: Victor Mataré <matare@fh-aachen.de>`
`Date:   Mon Apr 20 13:49:20 2020 +0200`
`systemd: call sleep with full path`

Thank you for very good help!

vmatare, 2020-04-21:
Great! Thanks for reporting the bug and helping to improve thinkfan!

closed 2020-04-21 by vmatare