vmatare/thinkfan #94: ERROR: Module thinkpad_acpi doesn't seem to support fan_control

arm3nyn, 2020-05-01:
I got this error when I do : sudo thinkfan -q
ERROR: ~TpFanDriver: Resetting fan control in /proc/acpi/ibm/fan: Invalid argument
ERROR: Module thinkpad_acpi doesn't seem to support fan_control

that mesans not supported for thinkpad e470?

vmatare, 2020-05-03:
Not necessarily, maybe you just haven't enabled the `fan_control` option on the `thinkpad_acpi` kernel module...

vmatare, 2020-05-03:
https://www.kernel.org/doc/html/latest/admin-guide/laptops/thinkpad-acpi.html

vmatare, 2020-05-04:
See also issue #45

vmatare, 2020-05-04:
Closing for now, since I highly suspect it's the same issue as in #45

closed 2020-05-04 by vmatare

mbernasocchi, 2020-07-15:
@vmatare I had enabled the `fan_control=1` and still had the error. adding èxperimental=1` removed the error and let thinkfan start. I put more details in #45