vmatare/thinkfan #95: Thinkstation P700

jorgelaranjo, 2020-05-20:
In reference to the closed item #52 

On a Thinkstation P700 folder `/proc/acpi/ibm/thermal` does not exist. `/sys/devices/platform/coretemp.*/` exist but those don't show any fans nor fan control whatsoever.
Lenovo does say this machine is [certified](https://support.lenovo.com/gb/en/solutions/linux-p700) for Linux [Ubuntu and CentOS] but neither seems to show fan control as my experience shows and the following link shows

Looking at Lenovo [website](https://pcsupport.lenovo.com/gb/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p700/downloads/driver-list/component?name=Software%20and%20Utilities) re drivers nothing relevant shows re Linux Fan control:

Extract from [Device Drivers update - DVD-ROM ISO image version for ThinkStation P500, P700, P900](https://download.lenovo.com/ibmdl/pub/pc/pccbbs/thinkcentre_drivers/readme_ts_p500p700p900_iso_files.txt)
```
Operating System   LINUX

  Software name                                                       Build ID      Version number
  ---------------------------------------------------------------     --------      -------------------------
  NVIDIA Video Card Driver					      1L3VDO104     346.59
  AMD Video Card Driver                                               1L2VDO084     14.502
  Realtek card reader(SD card reader on FPIO) driver                  1L7CAR014     1.06
  Realtek card reader(Taisol 29-in-1 card reader) driver              1L7CAR014     1.06
  Intel Xeon Phi Coprocessor Driver                                   1L1COP024     3.2.1
  LSI 9364-8i ROC Driver                                              1L8SAS067     6.808.09
  LSI IOC Mezzine Driver                                              1L8SAS067     6.808.09
  Fussion IO ioDrive                                                  1LHSSD034     3.2.8.1350
```

Would be nice to know if fan control is possible under Linux for this platform and which kernel modules need to be loaded.
`thinkpad_acpi `does not load.

vmatare, 2020-05-21:
Yeah, the `thinkpad_acpi` stuff is most probably specific to laptops, and the non-standard `/proc/acpi/ibm/*` interface is sort-of deprecated anyways. But there's a wealth of other hardware monitoring devices, many of which are supported by the Linux kernel out of the box, i.e. without any special driver updates. They are usually accessible through sysfs, i.e. somewhere under `/sys/class/hwmon`.

The easiest way to find out is to install the `lm-sensors` package and to then simply run `sensors`, which should list all hardware monitoring devices and their status:
```bash
sudo apt install lm-sensors
sensors
```

If that lists any fans with their current RPM, there's a good chance that driver also supports PWM fan control. Take a look in `/sys/class/hwmon` and see if any of the drivers have files named `pwm1` etc. in there.

If you can find no such thing, you might still have a chance because sometimes these hardware monitoring devices can't be auto-detected and the drivers have to be loaded manually. The `sensors-detect` tool (also from the `lm-sensors` package) might help you to find these. If that also doesn't work, take a look at your PCI bus with `sudo lspci` and see if there's anything interesting there. Post the output here if you're not sure what you're seeing.

jorgelaranjo, 2020-05-21:
`sensors` and `sensors-detect` don't get me there

Output of `sensors` after running `sensors-detect `and `'Y'` to all
```

coretemp-isa-0001
Adapter: ISA adapter
Package id 1: +42.0 C (high = +75.0 C, crit = +85.0 C)
Core 0: +35.0 C (high = +75.0 C, crit = +85.0 C)
Core 1: +33.0 C (high = +75.0 C, crit = +85.0 C)
Core 2: +37.0 C (high = +75.0 C, crit = +85.0 C)
Core 3: +35.0 C (high = +75.0 C, crit = +85.0 C)
Core 4: +35.0 C (high = +75.0 C, crit = +85.0 C)
Core 5: +33.0 C (high = +75.0 C, crit = +85.0 C)

nvme-pci-0200
Adapter: PCI adapter
Composite: +26.9 C (low = -273.1 C, high = +69.8 C)
(crit = +79.8 C)
Sensor 1: +26.9 C (low = -273.1 C, high = +65261.8 C)
Sensor 2: +26.9 C (low = -273.1 C, high = +65261.8 C)
Sensor 5: +34.9 C (low = -273.1 C, high = +65261.8 C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0: +40.0 C (high = +75.0 C, crit = +85.0 C)
Core 0: +35.0 C (high = +75.0 C, crit = +85.0 C)
Core 1: +34.0 C (high = +75.0 C, crit = +85.0 C)
Core 2: +32.0 C (high = +75.0 C, crit = +85.0 C)
Core 3: +32.0 C (high = +75.0 C, crit = +85.0 C)
Core 4: +32.0 C (high = +75.0 C, crit = +85.0 C)
Core 5: +32.0 C (high = +75.0 C, crit = +85.0 C)

coretemp-isa-0001
Adapter: ISA adapter
Package id 1:  +41.0 C  (high = +75.0 C, crit = +85.0 C)
Core 0:        +34.0 C  (high = +75.0 C, crit = +85.0 C)
Core 1:        +34.0 C  (high = +75.0 C, crit = +85.0 C)
Core 2:        +37.0 C  (high = +75.0 C, crit = +85.0 C)
Core 3:        +36.0 C  (high = +75.0 C, crit = +85.0 C)
Core 4:        +34.0 C  (high = +75.0 C, crit = +85.0 C)
Core 5:        +33.0 C  (high = +75.0 C, crit = +85.0 C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +26.9 C  (low  = -273.1 C, high = +69.8 C) (crit = +79.8 C)
Sensor 1:     +26.9 C  (low  = -273.1 C, high = +65261.8 C)
Sensor 2:     +26.9 C  (low  = -273.1 C, high = +65261.8 C)
Sensor 5:     +34.9 C  (low  = -273.1 C, high = +65261.8 C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +40.0 C  (high = +75.0 C, crit = +85.0 C)
Core 0:        +35.0 C  (high = +75.0 C, crit = +85.0 C)
Core 1:        +32.0 C  (high = +75.0 C, crit = +85.0 C)
Core 2:        +33.0 C  (high = +75.0 C, crit = +85.0 C)
Core 3:        +32.0 C  (high = +75.0 C, crit = +85.0 C)
Core 4:        +32.0 C  (high = +75.0 C, crit = +85.0 C)
Core 5:        +33.0 C  (high = +75.0 C, crit = +85.0 C)
```
`pwmconfig`
`/usr/bin/pwmconfig: There are no pwm-capable sensor modules installed`

`ls -al /sys/class/hwmon/`

```
lrwxrwxrwx  1 root root 0 May 21 08:50 hwmon0 -> ../../devices/pci0000:00/0000:00:01.1/0000:02:00.0/hwmon/hwmon0
lrwxrwxrwx  1 root root 0 May 21 08:50 hwmon1 -> ../../devices/platform/coretemp.0/hwmon/hwmon1
lrwxrwxrwx  1 root root 0 May 21 08:50 hwmon2 -> ../../devices/platform/coretemp.1/hwmon/hwmon2
```
[... 137 lines cut ...]
ff:14.2 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 0 ERROR Registers (rev 02)
ff:14.3 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 1 ERROR Registers (rev 02)
ff:14.6 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 0 & 1 (rev 02)
ff:14.7 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 0 & 1 (rev 02)
ff:15.0 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 2 Thermal Control (rev 02)
ff:15.1 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 3 Thermal Control (rev 02)
ff:15.2 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 2 ERROR Registers (rev 02)
ff:15.3 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 0 Channel 3 ERROR Registers (rev 02)
ff:16.0 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 1 Target Address, Thermal & RAS Registers (rev 02)
ff:16.6 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO Channel 2/3 Broadcast (rev 02)
ff:16.7 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO Global Broadcast (rev 02)
ff:17.0 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Integrated Memory Controller 1 Channel 0 Thermal Control (rev 02)
ff:17.4 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 2 & 3 (rev 02)
ff:17.5 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 2 & 3 (rev 02)
ff:17.6 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 2 & 3 (rev 02)
ff:17.7 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 DDRIO (VMSE) 2 & 3 (rev 02)
ff:1e.0 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Power Control Unit (rev 02)
ff:1e.1 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Power Control Unit (rev 02)
ff:1e.2 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Power Control Unit (rev 02)
ff:1e.3 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Power Control Unit (rev 02)
ff:1e.4 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 Power Control Unit (rev 02)
ff:1f.0 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 VCU (rev 02)
ff:1f.2 System peripheral: Intel Corporation Xeon E7 v3/Xeon E5 v3/Core i7 VCU (rev 02)

```

vmatare, 2020-05-21:
hm, usually there should be something behind the SMBus controller, but there might not be a Linux driver for it. Usually these system integrators deploy it with some halfway sensible default config and then they don't deem it necessary to support software control. Maybe you'll find some info on the chip in the detailed mainboard specifications. Then you could find out if there is a driver for some other, similar (e.g. older) chip from the same manufacturer and then contact the developer of that driver to extend support to your chip.

jorgelaranjo, 2020-05-21:
What I find strange is that Lenovo themselves say that the machine is Linux certified and none of the Certified versions shows fan control at all.

vmatare, 2020-05-21:
Yeah, one of the reasons I build my desktop systems from scratch. I get to choose a mainboard with properly documented components and I can make sure that the hardware monitoring chip is actually supported in Linux (i.e. with an *open source* driver in the mainline Linux kernel).

These certification programs tend to be a lot of smoke and mirrors. I wouldn't consider the system supported if you have to download some driver DVD. Usually what that means is their drivers are too crappy to qualify for inclusion into the Linux kernel, which means they'll only work with the horribly outdated Linux kernel they were certified for.

But sometimes you're lucky and open source development overtakes the legacy bubble created by the certification bullcrap. So if you're currently running Ubuntu LTS or RHEL/CentOS, I'd recommend trying out a distribution that isn't already outdated on release day, like Fedora, Arch or whatever. The current Linux kernel release is 5.6, and a big bunch of new hardware monitoring drivers came in a few releases ago. If you're still somewhere at 4.* there's a good chance that a more recent kernel might help you. Just make a USB boot stick with a recent Fedora and try out the lm-sensors stuff in there.

vmatare, 2020-05-21:
oh btw, I wouldn't think that this "fan speed" setting just sets a fixed fan speed. It should be a more or less agressive fan control curve that gives you more headroom before thermal throttling kicks in, at the price of more noise under load. So if you have that option definitely give it a try, it might do pretty much what you want without having to use thinkfan.

jorgelaranjo, 2020-05-21:
```
uname -r    
5.6.13-arch1-1
```

Fedora 32 live USB didn't help even after installing `lm-sensors`
The only difference was that in Fedora 32 I could see the Nvidia Fan Speed, nothing else really.
If the BIOS setting was the minimum speed that would be nice but what happens if you set it to a level and then after booting OS it goes to that level and stays there.
I may have to install Windows [can do on a spare partition/SSD] to check if Windows allows for control but I'd expect kernel 5.6 to offer control or Lenovo to supply drivers, to which is a professional certified machine.

jorgelaranjo, 2020-05-21:
Set fan level 3 out of 7 at BIOS/BOOT and loaded up CPU with stress test on 24 threads and all 12 cores [room temp 26C) operating at 2.6GHz
Fan never ever changed speed to bring temp down and let it go to the following:

```
coretemp-isa-0001
Adapter: ISA adapter
Package id 1:  +56.0 C  (high = +75.0 C, crit = +85.0 C)
Core 0:        +56.0 C  (high = +75.0 C, crit = +85.0 C)
Core 1:        +53.0 C  (high = +75.0 C, crit = +85.0 C)
Core 2:        +56.0 C  (high = +75.0 C, crit = +85.0 C)
Core 3:        +56.0 C  (high = +75.0 C, crit = +85.0 C)
Core 4:        +53.0 C  (high = +75.0 C, crit = +85.0 C)
Core 5:        +51.0 C  (high = +75.0 C, crit = +85.0 C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +24.9 C  (low  = -273.1 C, high = +69.8 C)
                       (crit = +79.8 C)
Sensor 1:     +24.9 C  (low  = -273.1 C, high = +65261.8 C)
Sensor 2:     +24.9 C  (low  = -273.1 C, high = +65261.8 C)
Sensor 5:     +32.9 C  (low  = -273.1 C, high = +65261.8 C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0 C  (high = +75.0 C, crit = +85.0 C)
Core 0:        +50.0 C  (high = +75.0 C, crit = +85.0 C)
Core 1:        +48.0 C  (high = +75.0 C, crit = +85.0 C)
Core 2:        +47.0 C  (high = +75.0 C, crit = +85.0 C)
Core 3:        +47.0 C  (high = +75.0 C, crit = +85.0 C)
Core 4:        +48.0 C  (high = +75.0 C, crit = +85.0 C)
Core 5:        +46.0 C  (high = +75.0 C, crit = +85.0 C)
```

at level 7 I know from exp CPU would be 40C at this stress level.

It would seam that fans are really set at a static level and that's that.

jorgelaranjo, 2020-05-21:
Taking into account all previously shared and the following [document](https://pcsupport.lenovo.com/us/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p710/solutions/ht500000) from Lenovo I strongly suspect that the fan speed is static and not minimum.

vmatare, 2020-05-21:
OK well that sounds pretty useless indeed. It may be worthwhile to talk to customer support, if only to make them aware that this is not adequate support for a high-end workstation. But maybe they can help you find out more about the hardware monitoring chip being used so you could be able to get the developer of a related driver to add support for your chip. In principle, you could also look over the mainboard and just google the labels on all the ICs to find it. But all of that is of course long shot. Depends on how much you really need active fan control.

jorgelaranjo, 2020-05-22:
> OK well that sounds pretty useless indeed. It may be worthwhile to talk to customer support, if only to make them aware that this is not adequate support for a high-end workstation. But maybe they can help you find out more about the hardware monitoring chip being used so you could be able to get the developer of a related driver to add support for your chip. In principle, you could also look over the mainboard and just google the labels on all the ICs to find it. But all of that is of course a long shot. Depends on how much you really need active fan control.

It is not so much that I need *active control* of fan speed but that I'd like the machine to have dynamic control of its fans, something I cannot see happening, thus the need for *active control*

vmatare, 2020-07-31:
Hey, I just stumbled across this: https://github.com/hirschmann/nbfc
It's a cross-platform solution that basically bypasses any sensor drivers and talks directly to the hardware. Seems like the community also (by necessity) has a lot of experience in figuring out weird hardware details. Maybe they can help you.

mrhpearson, 2021-02-10:
Hi - I got pointed at this and I can't answer for the platforms listed but one of my team got fan control working on the P720 and that might be useful as a starting point?

On the P720 control is done via the NCT6685D eSIO chip, so the nct6683 driver is needed. However when you try and load that you get a bunch of ACPI warnings about SystemIO range. You can bypass this by adding "acpi_enforce_resources=lax" to the kernel bootargs.

We then had an updated /etc/sensors3.conf file (attached as a .txt file)
[sensors3.txt](https://github.com/vmatare/thinkfan/files/5960146/sensors3.txt)

Then the sensors command worked....and I think you should be able to do fan control.

Let me know if it helps - I'm guessing completely that the same eSIO chip is being used so the above might all be useless.
Mark

jorgelaranjo, 2021-04-01:
> Hi - I got pointed at this and I can't answer for the platforms listed but one of my team got fan control working on the P720 and that might be useful as a starting point?
>
> On the P720 control is [...]

On the P700 as it stand `sensors-detect` brings
```
Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)
```

Has `sensors-detect` on the P720 gave the NCT6685D as output?

jorgelaranjo, 2021-05-13:
> Hi - I got pointed at this and I can't answer for the platforms listed but one of my team got fan control working on the P720 and that might be useful as a starting point?
>
> On the P720 control is [...]

How do you then control the fans? Using Thinkfan?